esp-rs/rust #4: NVS and `compiler_builtins` collision

lexxvir, 2019-08-02:
Hi @MabezDev,

It is not issue of your fork, but I just want to publish my experience with NVS (Non Volatile Storage of ESP32). I think we could track it here.

ESP IDF has such component as NVS. It is key-value storage built upon flash. Any action with NVS translates to flash interactions. Because application runs also from flash, IDF internally use some locks, to prevent flash interaction until NVS operation is done. Also, NVS uses some low level functions such `memcpy`. Such functions in IDF marked as `IRAM`, so they are placed in the Internal RAM, not in flash. So, locks and `IRAM` functions allows NVS does things while application runs from flash.

When we compiling rust application, rust compiler implicitly add `libcompiler_builtins` library to the linker parameters. `libcompiler_builtins` also contains low level functions such as `memcpy`, but these functions don't marked as `IRAM`. Because `libcompiler_builtins` appears first in linker arguments, linker uses `memcpy` from `libcompiler_builtins` instead of right version from Espressif's `libc`.

This issue appears as `Guru Meditation Error: Core N panic'ed (IllegalInstruction)` when you try to call `nvs_flash_init()` from IDF. Unfortunately, NVS is required for normal work of builtin WiFI, it is too hard to use WiFi without NVS.

pacmancoder, 2019-08-10:
@lexxvir I think I found some temporary solution:

If you are targeting your rust code as static library, it can be processed with objdump to rename all conflicting symbols:

Here what I have done for my esp8266 project:
```bash
xtensa-lx106-elf-objcopy \
        --redefine-syms=symbol_renaming.map \
        firmware/target/xtensa-esp8266-none-elf/release/libfirmware.a
```

Where libfirmware.a is my resulting static library, which I link inside IDF shim project.

symbol_renaming.map has the following contents:
```
__ledf2 __rust__ledf2
__gedf2 __rust__gedf2
__unorddf2 __rust__unorddf2
__eqdf2 __rust__eqdf2
__ltdf2 __rust__ltdf2
__nedf2 __rust__nedf2
__gtdf2 __rust__gtdf2
__modsi3 __rust__modsi3
```

Example project can be found at (currently at rust-migration branch):
https://github.com/pacmancoder/shelter-unicorn/tree/migration-rust

But still, it is a pretty questionable workaround :smile:

lexxvir, 2019-08-11:
Hi @pacmancoder,

Thank you for share you workaround!

Unfortunately, I do not embedded Rust static library into C code, contrarily I compile regular Rust crate (lib/bin) that uses C code internally (IDF and couple of existing libraries). So I have not stage where I can remap symbol names.

Currently I use dumb workaround. I redefined linker used by the Rust compiler (through `.cargo/config`) to shell script. The script cuts `libcompiler_builtin.a` out from the linker command line parameters and calls real linker (g++). So far I have not issues with such solution, but I admit it is ugly hack.

RomanSamusevich, 2019-08-12:
Hi @lexxvir 
I think it depends on your work environment. I have not found this problem. compiler_builtins works fine by default.
I know that because of printf different errors may appear (I still do not quite understand why).
I use this code for output:
```
extern "C"{
    pub fn write(fd: i32, buf: *const core::ffi::c_void, count: u32) -> u32;
}

#[inline]
fn libc_println(handle: u32, msg: &str){
    unsafe {
        write(
            handle as i32,
            msg.as_ptr() as *const core::ffi::c_void,
            msg.len() as u32,
        );
    }
}
```

lexxvir, 2019-08-13:
Hi @RomanSamusevich,

Do you use NVS? This issue appears only when you call NVS related function from IDF.

RomanSamusevich, 2019-08-13:
Hi @lexxvir
All nvs functions (nvs.h, nvs_flash.h) work for me (wifi has not tested yet).  I use the latest MabezDev / rust-xtensa, compiler-builtins 0.1.19 (standard unchanged).  I use cargo xbuild for assembly.

lexxvir, 2019-08-31:
@romevich 

Could please dump linker arguments when you linking your crate ?

romevich, 2019-09-04:
@lexxvir 

cargo config

```
[build]
target = "xtensa-esp32-none-elf"

[target.xtensa-esp32-none-elf]
rustflags = [
  "-C", "save-temps",
  "-C", "link-arg=-nostartfiles",
  "-C", "target-cpu=esp32",
  "-C", "link-arg=-nostdlib",
  "-C", "link-arg=-ucall_user_start_cpu0",
  "-C", "link-arg=-Wl,--gc-sections",
  "-C", "link-arg=-Wl,-static",
  "-C", "link-arg=-Wl,--start-group",
  "-C", "link-arg=-L/home/roman/Experimental/rust/rust1/rust_test/blink/idf_project/build/app_trace",
  "-C", "link-arg=-lapp_trace",
  "-C", "link-arg=-L/home/roman/Experimental/rust/rust1/rust_test/blink/idf_project/build/app_update",
  "-C", "link-arg=-lapp_update",
  "-C", "link-arg=-uesp_app_desc",
  #"-C", "link-arg=-L/home/roman/Experimental/rust/rust1/rust_test/blink/idf_project/build/asio",
  #"-C", "link-arg=-lasio",
  "-C", "link-arg=-L/home/roman/Experimental/rust/rust1/rust_test/blink/idf_project/build/bootloader_support",
  "-C", "link-arg=-lbootloader_support",
  "-C", "link-arg=-L/home/roman/Experimental/rust/rust1/rust_test/blink/idf_project/build/bt",
  "-C", "link-arg=-lbt",
  "-C", "link-arg=-L/home/roman/Experimental/rust/rust1/rust_test/blink/idf_project/build/coap",
  "-C", "link-arg=-lcoap",
  "-C", "link-arg=-L/home/roman/Experimental/rust/rust1/rust_test/blink/idf_project/build/console",
  "-C", "link-arg=-lconsole",
  "-C", "link-arg=-L/home/roman/Experimental/rust/rust1/rust_test/blink/idf_project/build/driver",
  "-C", "link-arg=-ldriver",
  "-C", "link-arg=-L/home/roman/Experimental/rust/rust1/rust_test/blink/idf_project/build/efuse",
  "-C", "link-arg=-lefuse",
  "-C", "link-arg=-L/home/roman/Experimental/rust/rust1/rust_test/blink/idf_project/build/esp-tls",
  "-C", "link-arg=-lesp-tls",
  "-C", "link-arg=-L/home/roman/Experimental/rust/rust1/rust_test/blink/idf_project/build/esp32",
  "-C", "link-arg=-lesp32",
  "-C", "link-arg=-L/home/roman/Experimental/esp/esp-idf/components/esp32/ld",
  "-C", "link-arg=-Tesp32_out.ld",
  "-C", "link-arg=-uld_include_panic_highint_hdl",
  "-C", "link-arg=-T/home/roman/Experimental/rust/rust1/rust_test/blink/idf_project/build/esp32/esp32.project.ld",
  "-C", "link-arg=-Tesp32.peripherals.ld",
  "-C", "link-arg=-L/home/roman/Experimental/rust/rust1/rust_test/blink/idf_project/build/esp_adc_cal",
  "-C", "link-arg=-lesp_adc_cal",
  "-C", "link-arg=-L/home/roman/Experimental/rust/rust1/rust_test/blink/idf_project/build/esp_common",
  "-C", "link-arg=-lesp_common",
  "-C", "link-arg=-L/home/roman/Experimental/rust/rust1/rust_test/blink/idf_project/build/esp_eth",
  "-C", "link-arg=-lesp_eth",
  "-C", "link-arg=-L/home/roman/Experimental/rust/rust1/rust_test/blink/idf_project/build/esp_event",
  "-C", "link-arg=-lesp_event",
  "-C", "link-arg=-L/home/roman/Experimental/rust/rust1/rust_test/blink/idf_project/build/esp_gdbstub",
  "-C", "link-arg=-lesp_gdbstub",
  "-C", "link-arg=-L/home/roman/Experimental/rust/rust1/rust_test/blink/idf_project/build/esp_http_client",
  "-C", "link-arg=-lesp_http_client",
  "-C", "link-arg=-L/home/roman/Experimental/rust/rust1/rust_test/blink/idf_project/build/esp_http_server",
  "-C", "link-arg=-lesp_http_server",
  "-C", "link-arg=-L/home/roman/Experimental/rust/rust1/rust_test/blink/idf_project/build/esp_https_ota",
  "-C", "link-arg=-lesp_https_ota",
  "-C", "link-arg=-L/home/roman/Experimental/rust/rust1/rust_test/blink/idf_project/build/esp_local_ctrl",
  "-C", "link-arg=-lesp_local_ctrl",
  "-C", "link-arg=-L/home/roman/Experimental/rust/rust1/rust_test/blink/idf_project/build/esp_ringbuf",
  "-C", "link-arg=-lesp_ringbuf",
  "-C", "link-arg=-L/home/roman/Experimental/rust/rust1/rust_test/blink/idf_project/build/esp_rom",
  "-C", "link-arg=-lesp_rom",
  "-C", "link-arg=-L/home/roman/Experimental/esp/esp-idf/components/esp_rom/esp32/ld",
  "-C", "link-arg=-Tesp32.rom.ld",
[... 81 lines cut ...]
  "-C", "link-arg=-ltcp_transport",
  "-C", "link-arg=-L/home/roman/Experimental/rust/rust1/rust_test/blink/idf_project/build/tcpip_adapter",
  "-C", "link-arg=-ltcpip_adapter",
  "-C", "link-arg=-L/home/roman/Experimental/rust/rust1/rust_test/blink/idf_project/build/ulp",
  "-C", "link-arg=-lulp",
  #"-C", "link-arg=-L/home/roman/Experimental/rust/rust1/rust_test/blink/idf_project/build/unity",
  #"-C", "link-arg=-lunity",
  "-C", "link-arg=-L/home/roman/Experimental/rust/rust1/rust_test/blink/idf_project/build/vfs",
  "-C", "link-arg=-lvfs",
  "-C", "link-arg=-L/home/roman/Experimental/rust/rust1/rust_test/blink/idf_project/build/wear_levelling",
  "-C", "link-arg=-lwear_levelling",
  "-C", "link-arg=-L/home/roman/Experimental/rust/rust1/rust_test/blink/idf_project/build/wifi_provisioning",
  "-C", "link-arg=-lwifi_provisioning",
  "-C", "link-arg=-L/home/roman/Experimental/rust/rust1/rust_test/blink/idf_project/build/wpa_supplicant",
  "-C", "link-arg=-lwpa_supplicant",
  "-C", "link-arg=-L/home/roman/Experimental/rust/rust1/rust_test/blink/idf_project/build/xtensa",
  "-C", "link-arg=-lxtensa",
  "-C", "link-arg=/home/roman/Experimental/esp/esp-idf/components/xtensa/esp32/libhal.a",
  "-C", "link-arg=-lgcc",
  "-C", "link-arg=-lstdc++",
  "-C", "link-arg=-lgcov",
  "-C", "link-arg=-Wl,--end-group",
  "-C", "link-arg=-Wl,-EL",
]
```

lexxvir, 2019-09-04:
@romevich Sorry, I mean arguments that _actually_ passed to the linker, not those you have configured.

Most easiest way to get all arguments it is define somewhere non existed function and call it:
```
#![no_std]
#![no_main]

use core::panic::PanicInfo;

extern "C" {
    fn non_exist_func();
}

#[panic_handler]
fn panic(_info: &PanicInfo) -> ! {
    loop {}
}

#[no_mangle]
unsafe fn app_main() {
    non_exist_func();
}
```

So you should get linker error with full command line arguments (explicit and implicit ones).

Or you can redefine linker in `.cargo/config` to shell script and just dump passed arguments to file.

romevich, 2019-09-05:
@lexxvir 

```
error: linking with `xtensa-esp32-elf-gcc` failed: exit code: 1
  |
  = note: "xtensa-esp32-elf-gcc" "-L" "/home/roman/Experimental/rust/rust1/rust_test/blink/target/sysroot/lib/rustlib/xtensa-esp32-none-elf/lib" "/home/roman/Experimental/rust/rust1/rust_test/blink/target/xtensa-esp32-none-elf/release/deps/blink-6db9f1d396fe6254.blink.cnqv2jox-cgu.0.rcgu.o" "-o" "/home/roman/Experimental/rust/rust1/rust_test/blink/target/xtensa-esp32-none-elf/release/deps/blink-6db9f1d396fe6254" "-Wl,--gc-sections" "-nodefaultlibs" "-L" "/home/roman/Experimental/rust/rust1/rust_test/blink/target/xtensa-esp32-none-elf/release/deps" "-L" "/home/roman/Experimental/rust/rust1/rust_test/blink/target/release/deps" "-L" "/home/roman/Experimental/rust/rust1/rust_test/blink/target/sysroot/lib/rustlib/xtensa-esp32-none-elf/lib" "-Wl,-Bstatic" "/home/roman/Experimental/rust/rust1/rust_test/blink/target/sysroot/lib/rustlib/xtensa-esp32-none-elf/lib/librustc_std_workspace_core-f651658f9c3c2a0a.rlib" "/home/roman/Experimental/rust/rust1/rust_test/blink/target/sysroot/lib/rustlib/xtensa-esp32-none-elf/lib/libcore-f79f47342aa3c2d5.rlib" "/home/roman/Experimental/rust/rust1/rust_test/blink/target/sysroot/lib/rustlib/xtensa-esp32-none-elf/lib/libcompiler_builtins-0686d155b28882ab.rlib" "-nostartfiles" "-nostdlib" "-ucall_user_start_cpu0" "-Wl,--gc-sections" "-Wl,-static" "-Wl,--start-group" "-L/home/roman/Experimental/rust/rust1/rust_test/blink/idf_project/build/app_trace" "-lapp_trace" "-L/home/roman/Experimental/rust/rust1/rust_test/blink/idf_project/build/app_update" "-lapp_update" "-uesp_app_desc" "-L/home/roman/Experimental/rust/rust1/rust_test/blink/idf_project/build/bootloader_support" "-lbootloader_support" "-L/home/roman/Experimental/rust/rust1/rust_test/blink/idf_project/build/bt" "-lbt" "-L/home/roman/Experimental/rust/rust1/rust_test/blink/idf_project/build/coap" "-lcoap" "-L/home/roman/Experimental/rust/rust1/rust_test/blink/idf_project/build/console" "-lconsole" "-L/home/roman/Experimental/rust/rust1/rust_test/blink/idf_project/build/driver" "-ldriver" "-L/home/roman/Experimental/rust/rust1/rust_test/blink/idf_project/build/efuse" "-lefuse" "-L/home/roman/Experimental/rust/rust1/rust_test/blink/idf_project/build/esp-tls" "-lesp-tls" "-L/home/roman/Experimental/rust/rust1/rust_test/blink/idf_project/build/esp32" "-lesp32" "-L/home/roman/Experimental/esp/esp-idf/components/esp32/ld" "-Tesp32_out.ld" "-uld_include_panic_highint_hdl" "-T/home/roman/Experimental/rust/rust1/rust_test/blink/idf_project/build/esp32/esp32.project.ld" "-Tesp32.peripherals.ld" "-L/home/roman/Experimental/rust/rust1/rust_test/blink/idf_project/build/esp_adc_cal" "-lesp_adc_cal" "-L/home/roman/Experimental/rust/rust1/rust_test/blink/idf_project/build/esp_common" "-lesp_common" "-L/home/roman/Experimental/rust/rust1/rust_test/blink/idf_project/build/esp_eth" "-lesp_eth" "-L/home/roman/Experimental/rust/rust1/rust_test/blink/idf_project/build/esp_event" "-lesp_event" "-L/home/roman/Experimental/rust/rust1/rust_test/blink/idf_project/build/esp_gdbstub" "-lesp_gdbstub" "-L/home/roman/Experimental/rust/rust1/rust_test/blink/idf_project/build/esp_http_client" "-lesp_http_client" "-L/home/roman/Experimental/rust/rust1/rust_test/blink/idf_project/build/esp_http_server" "-lesp_http_server" "-L/home/roman/Experimental/rust/rust1/rust_test/blink/idf_project/build/esp_https_ota" "-lesp_https_ota" "-L/home/roman/Experimental/rust/rust1/rust_test/blink/idf_project/build/esp_local_ctrl" "-lesp_local_ctrl" "-L/home/roman/Experimental/rust/rust1/rust_test/blink/idf_project/build/esp_ringbuf" "-lesp_ringbuf" "-L/home/roman/Experimental/rust/rust1/rust_test/blink/idf_project/build/esp_rom" "-lesp_rom" "-L/home/roman/Experimental/esp/esp-idf/components/esp_rom/esp32/ld" "-Tesp32.rom.ld" "-Tesp32.rom.libgcc.ld" "-Tesp32.rom.syscalls.ld" "-Tesp32.rom.newlib-data.ld" "-Tesp32.rom.newlib-funcs.ld" "-L/home/roman/Experimental/rust/rust1/rust_test/blink/idf_project/build/esp_websocket_client" "-lesp_websocket_client" "-L/home/roman/Experimental/rust/rust1/rust_test/blink/idf_project/build/esp_wifi" "-lesp_wifi" "-L/home/roman/Experimental/esp/esp-idf/components/esp_wifi/lib_esp32" "-lcore" "-lrtc" "-lnet80211" "-lpp" "-lsmartconfig" "-lcoexist" "-lespnow" "-lphy" "-lmesh" "-L/home/roman/Experimental/rust/rust1/rust_test/blink/idf_project/build/espcoredump" "-lespcoredump" "-L/home/roman/Experimental/rust/rust1/rust_test/blink/idf_project/build/fatfs" "-lfatfs" "-L/home/roman/Experimental/rust/rust1/rust_test/blink/idf_project/build/freemodbus" "-lfreemodbus" "-L/home/roman/Experimental/rust/rust1/rust_test/blink/idf_project/build/freertos" "-lfreertos" "-Wl,--undefined=uxTopUsedPriority" "-L/home/roman/Experimental/rust/rust1/rust_test/blink/idf_project/build/heap" "-lheap" "-L/home/roman/Experimental/rust/rust1/rust_test/blink/idf_project/build/jsmn" "-ljsmn" "-L/home/roman/Experimental/rust/rust1/rust_test/blink/idf_project/build/json" "-ljson" "-L/home/roman/Experimental/rust/rust1/rust_test/blink/idf_project/build/libsodium" "-llibsodium" "-L/home/roman/Experimental/rust/rust1/rust_test/blink/idf_project/build/log" "-llog" "-L/home/roman/Experimental/rust/rust1/rust_test/blink/idf_project/build/lwip" "-llwip" "-L/home/roman/Experimental/rust/rust1/rust_test/blink/idf_project/build/main" "-lmain" "-L/home/roman/Experimental/rust/rust1/rust_test/blink/idf_project/build/mbedtls" "-lmbedtls" "-L/home/roman/Experimental/rust/rust1/rust_test/blink/idf_project/ [... cut ...] lgcc" "-lstdc++" "-lgcov" "-Wl,--end-group" "-Wl,-EL" "-Wl,-Bdynamic"
  = note: /home/roman/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/roman/Experimental/rust/rust1/rust_test/blink/target/xtensa-esp32-none-elf/release/deps/blink-6db9f1d396fe6254.blink.cnqv2jox-cgu.0.rcgu.o:(.literal.literal+0x0): undefined reference to `phantom_func'
          collect2: error: ld returned 1 exit status
```

lexxvir, 2019-09-06:
@romevich It seems you are using ESP IDF v.4.0 (or master branch). I use stable release v.3.3. There is chance that NVS implementation was changed and it fixed issue somehow.

I will try to migrate to IDF 4.0 and check issue again.

pacmancoder, 2019-09-06:
@lexxvir Forgot to mention, I have got this issue on the master branch for esp8266 RTOS SDK (IDF-style).

reitermarkus, 2019-12-11:
I'm using ESP-IDF 4.0-beta2 and there don't seem to be any conflicting symbols with NVS.

MabezDev, 2020-04-07:
Hope this is all sorted now. Let me know if it still persists.

closed 2020-04-07 by MabezDev