esp-rs/rust #7: Update to newer compiler version.

0ndorio, 2019-09-05:
@MabezDev  Your repo is still based on  `rustc v1.33.0` but your patches apply without any complications on stable `rustc v1.37.0` (maybe even current beta or nightly).

What do you think about rebasing or merging onto a newer version? Especially as `rustc v1.36.0` stabilized the `alloc` crate which means that any `no_sdt` crate which uses `extern crate alloc` without the `#[feature(alloc)]` flag can't get with the current version.

I did a few tests locally with an updated version and couldn't trigger any issues. Is there anything related to `llvm-xtensa` which would require to stay on an older version?

MabezDev, 2019-09-05:
Oh really? I was under the impression the minimum llvm version for rust was bumped since, if that's not the case I'd happily rebase.

0ndorio, 2019-09-05:
You may do a few tests before but at least for me it worked out. For example `https://github.com/cuviper/num-bigint/tree/alloc`  (an open merge request on `num-bigint` for `no_std` environments) couldn't be compiled before not it works just fine.

MabezDev, 2019-09-05:
Looked into it a bit further, [as per the 1.33 change log](https://github.com/rust-lang/rust/blob/master/RELEASES.md#compiler-3) the minimum supported llvm version is 6. So it should be fine to rebase. I'll try to get to it this weekend.

pacmancoder, 2019-09-16:
Hello! So, it there are any news on rebasing rust-xtensa to 1.36/1.37 ? 😃

0ndorio, 2019-09-17:
@pacmancoder If you want to try it you could temporary use https://github.com/0ndorio/rust-xtensa/commit/dfad57c025b9206ebbba6691ad74746779ab00c9. This is a merge of stable 1.37 without any merge conflicts during the process.

MabezDev, 2019-09-17:
Sorry this took so long! I've got a PR open with the rebase, I'd like to give it few days for some people to test and chime in on any issue before merging.

MabezDev, 2019-09-18:
Closed in #8

closed 2019-09-18 by MabezDev