esp-rs/rust #13: Crash dropping `Arc`

rodrigorc, 2019-11-22:
I've had some crashes when dropping `core::sync::Arc` objects. After some debugging I tracked the issue down to a LLVM bug with the atomic substraction, that I reported [here](https://github.com/espressif/llvm-xtensa/issues/12), but not reply yet :-(.

For now I worked around the LLVM bug with this hack:
```
diff --git a/src/libcore/sync/atomic.rs b/src/libcore/sync/atomic.rs
index 8dfb19fa032..eb59616ada0 100644
--- a/src/libcore/sync/atomic.rs
+++ b/src/libcore/sync/atomic.rs
@@ -1577,7 +1577,8 @@ assert_eq!(foo.load(Ordering::SeqCst), 10);
                 #[$stable]
                 #[cfg(target_has_atomic = "cas")]
                 pub fn fetch_sub(&self, val: $int_type, order: Ordering) -> $int_type {
-                    unsafe { atomic_sub(self.v.get(), val, order) }
+                    unsafe { atomic_add(self.v.get(), val.wrapping_neg(), order) }
                 }
             }
```
I admit that this is not pretty, I'd rather modify `atomic_sub()` itself, but it is too generic to fix it easily.

MabezDev, 2019-11-26:
Thanks for posting here, I'm sure many will find this helpful, until it is fixed.

rodrigorc, 2019-11-26:
Would you take a pull request with this patch, or maybe another similar hack to minimize the impact of this LLVM bug? I think it would ease the adoption Rust for the ESP32. as `Arc` is a basic building block for many embedded projecs.

MabezDev, 2020-01-18:
Could you try #20 and see if it's fixed?

reitermarkus, 2020-01-26:
It's working without the workaround now.

closed 2020-04-07 by MabezDev