esp-rs/rust #18: Verify the Rust ABI impl against the LLVM ABI impl

MabezDev, 2020-01-17:
We should make sure that [our ABI implementation](https://github.com/MabezDev/rust-xtensa/blob/xtensa-target/src/librustc_target/abi/call/xtensa.rs) performs identically to the [llvm implementation](https://github.com/espressif/llvm-project/blob/36ac5025f4346a01213995bdba591a6ebb513e50/clang/lib/CodeGen/TargetInfo.cpp#L9686), not including vaargs as Rust does not support these.

vmarkushin, 2020-04-12:
Here is information about debugging rustc's type layouts: https://www.ralfj.de/blog/2020/04/04/layout-debugging.html

MabezDev, 2022-08-11:
The Rust ABI is wrong, it needs to be updated. Reference for how it should be done is here: https://github.com/espressif/llvm-project/blob/e9f57cdbcf3e0a63f395e683ccfaf7c4e6e1b093/clang/lib/CodeGen/TargetInfo.cpp#L11241

ollpu, 2022-08-11:
Structs seem to be incorrectly split between registers and stack, when single arguments should be exclusively kept in one of them according to the ISA manual:

> Each argument must be passed entirely in registers or entirely on the stack; an argument cannot be split with some words in registers and the remainder on the stack. If an
argument does not fit entirely in the remaining unused registers, it is passed on the
stack and those registers remain unused.

To demonstrate the disparity, C-code built as a custom component with `esp-idf-sys` (which to my understanding uses GCC):

```c
struct Pair {
  int a, b;
};

void abi_test_5(int p1, int p2, int p3, int p4, struct Pair p5) {
    ESP_LOGW("abi_test_5", "p5.a=%d, p5.b=%d", p5.a, p5.b);
}

void abi_test_6(int p1, int p2, int p3, int p4, int p5, struct Pair p6) {
    ESP_LOGW("abi_test_6", "p6.a=%d, p6.b=%d", p6.a, p6.b);
}
```
Rust code:
```rust
use esp_idf_sys::c_types::c_int;

#[repr(C)]
struct Pair {
    a: c_int,
    b: c_int,
}

extern "C" {
    fn abi_test_5(p1: c_int, p2: c_int, p3: c_int, p4: c_int, p5: Pair);
    fn abi_test_6(p1: c_int, p2: c_int, p3: c_int, p4: c_int, p5: c_int, p6: Pair);
}

fn main() {
    unsafe {
        abi_test_5(0, 0, 0, 0, Pair { a: 1, b: 2 });
        abi_test_6(0, 0, 0, 0, 0, Pair { a: 1, b: 2 });
    }
}
```
outputs:
```
W (1335) abi_test_5: p5.a=1, p5.b=2
W (1335) abi_test_6: p6.a=2, p6.b=12288
```

MabezDev, 2022-08-16:
@ollpu I'm working on this now, could you provide your test case as a repo?

ollpu, 2022-08-17:
@MabezDev Here you go: https://github.com/ollpu/esp-rustc-abi-bug. Compiles, but I can't test it on hardware right now.

MabezDev, 2022-08-18:
Thanks for the test cases @ollpu! I extended them a little more here: https://github.com/MabezDev/esp-rustc-abi-bug/. I think I have the ABI fixed up in https://github.com/esp-rs/rust/commit/564a0ce80e6837e3a90c5b8adf9ea15a2beff64e.

I'll see if we can get a patch release of our compiler out soon. In the meantime, and if you are able, could you try out the branch above? It would require compiling the rustc, so I understand if you don't want to :D

ollpu, 2022-08-19:
@MabezDev Looking good!

```
I (397) abi_test_5: p5.a=1, p5.b=2
I (397) abi_test_6: p6.a=1, p6.b=2
I (407) abi_test_7: p7.a=1, p7.b=2
I (407) abi_test_8: p8.a=1, p8.b=2
I (417) abi_test_18: p18.a=1, p18.b=2
I (417) ret_abi_test_internal_4: a=1, b=2, c=3, d=4
I (427) ret_abi_test_internal_5: a=1, b=2, c=3, d=4, e=5
ret_abi_test_4: Return4 { a: 1, b: 2, c: 3, d: 4 }
ret_abi_test_5: Return5 { a: 1, b: 2, c: 3, d: 4, e: 5 }
```

MabezDev, 2022-08-19:
Thanks, @ollpu! I've just pushed to the 1.63.0.1 release branch with this fix included. Builds should be available soon :)

closed 2022-08-19 by MabezDev

georgik, 2022-08-19:
Artifacts will become available here: https://github.com/esp-rs/rust-build/releases/tag/v1.63.0.1

ollpu, 2022-08-24:
Heads up: We noticed that between updating from `1.63.0.0` to `1.63.0.1`, `std::thread::sleep` no longer works – it doesn't wait at all. I didn't have time to comb through the call tree yet, but this could be indicative of an ABI related miscompilation.