esp-rs/rust #27: ESP8266 build regression

vmarkushin, 2020-04-07:
Hi,

After today's update, I started getting this compilation error of 'quickstart' project:
```
$ cargo xbuild      
<...>
LLVM ERROR: Error while trying to spill A8 from class AR: Cannot scavenge register without an emergency spill slot!
error: could not compile `core`.
```

...and another error when building with `xargo`:
```
$ xargo build
error: cannot produce bin for `esp32-test v0.1.0 (/Users/vmark/dev/emb/xtensa-rust-quickstart)` as the target `xtensa-esp8266-none-elf` does not support these crate types
```

Rustc info:
```
rustc 1.44.0-dev
binary: rustc
commit-hash: unknown
commit-date: unknown
host: x86_64-apple-darwin
release: 1.44.0-dev
LLVM version: 9.0
```

Note, that with `xtensa-esp32-none-elf` target the project compiles successfully. 

Should I report the issue on espressif's LLVM fork?

MabezDev, 2020-04-09:
Thanks for the report! The current ABI in the `rust-xtensa`, is based of the old forks ABI. It's quite possible that, this ABI implementation is now wrong, especially for the esp8266 as this gets tested alot less than the esp32. I opened https://github.com/MabezDev/rust-xtensa/issues/18 previously as I was concerned this might be the case.

Unfortunately, I've had a lot on my plate recently, and haven't had a chance to review. Hopefully I can take a look soon, or if you wish, it would be great to get someone else's eyeballs on it!

zinahe, 2020-05-20:
@vmarkushin,

Please see below regarding the second error:

https://github.com/MabezDev/rust-xtensa/issues/29#issuecomment-627683708

After that, I was able to build and run a "blinky" on ESP8266.

Cheers,

-Zinahe A.

vmarkushin, 2020-06-13:
@zinahe 
This kinda helped me. Also, switched from `xargo` to `cargo-xbuild` to successfully build the project.

closed 2020-06-13 by vmarkushin