esp-rs/rust #38: Structure alignment !=4 bytes

arjanmels, 2020-07-18:
When creating structures with u8 or u16 members, these are not aligned to 4 byte boundaries.

This leads to alignment exceptions when they are stored in ROM (or external memory).

I am not certain if it can be changed via the data-layout in the rust compiler or if it needs to be addressed in llvm.

Alternative is to implemented the alignment exception handler. However this will be slow.

@MabezDev, @andreisfr Any thoughts on this?

arjanmels, 2020-07-18:
If I change the data_layout string in xtensa_esp32_none_elf.rs to: "e-m:e-p:32:32-i8:32-i16:32-i64:64-n32".
I get compile errors:

```
error[E0080]: it is undefined behavior to use this value
   --> /home/mels/programs/rust-xtensa/src/libcore/fmt/num.rs:183:1
    |
183 | / static DEC_DIGITS_LUT: &[u8; 200] = b"0001020304050607080910111213141516171819\
184 | |       2021222324252627282930313233343536373839\
185 | |       4041424344454647484950515253545556575859\
186 | |       6061626364656667686970717273747576777879\
187 | |       8081828384858687888990919293949596979899";
    | |________________________________________________^ type validation failed: encountered an unaligned reference (required 4 byte alignment but found 1)
    |
    = note: The rules on what exactly is undefined behavior aren't clear, so this check might be overzealous. Please open an issue on the rustc repository if you believe it should not be considered undefined behavior.

error[E0080]: it is undefined behavior to use this value
    --> /home/mels/programs/rust-xtensa/src/libcore/fmt/mod.rs:1432:21
     |
1432 | /                     const ZEROES: &str = // 64 zeroes
1433 | |                         "0000000000000000000000000000000000000000000000000000000000000000";
     | |___________________________________________________________________________________________^ type validation failed: encountered an unaligned reference (required 4 byte alignment but found 1)
     |
     = note: The rules on what exactly is undefined behavior aren't clear, so this check might be overzealous. Please open an issue on the rustc repository if you believe it should not be considered undefined behavior.
```

Gasper, 2020-07-18:
I think you'll probably want something like this: e-m:e-p:32:32-i1:8:32-i8:8:32-i16:16:32-i64:64-f64:64-a:0:32-n32
as in i<integer size>:<size>:<alignment>
See this: https://llvm.org/docs/LangRef.html#data-layout

An if I understand correctly, the 'a' key for aggregate should align the structures

Besides the file you're editing I think you'll get runtime error if it doesn't match the LLVM data layout code: https://github.com/MabezDev/llvm-project/blob/xtensa_release_9.0.1_with_rust_patches/llvm/lib/Target/Xtensa/XtensaTargetMachine.cpp#L34
So you'll probably have to edit both.

arjanmels, 2020-07-18:
@Gasper Thanks, I am giving it a try. However, I think it should be a:32:32, to always force 32 bit alignment isn't it? 
My understanding of a:\<abi\>:\<pref\>, is <abi> is the absolute minimum, <pref> is preferred.

Gasper, 2020-07-18:
Ah, yes, you're right. I guess the idea of preferred alignment is to have 4B alignment most of the time and exception handler for packed structures and similar cases...

And it also makes sense that you get that compile error, since those `u8`s are 1B-aligned.

arjanmels, 2020-07-19:
Oeps it seems, that the struct was not as literal placed in the text segment, but that I made a mistake in the link script.

Closing this issue

closed 2020-07-19 by arjanmels