esp-rs/rust #41: Add a builder in docker

tlightsky, 2020-10-27:
so that any one can build the target only depend on docker,
(like this one: https://github.com/marcelstoer/docker-nodemcu-build)

jessebraham, 2020-10-27:
I have written this `Dockerfile` for this repository that you can try out if you wish. If there's interest it could be added here.
https://github.com/jessebraham/docker-rust-esp

jeikabu, 2021-02-11:
There's quite a few of these floating around:

- https://github.com/mtnmts/rust-esp32-container
- Based on https://github.com/ctron/rust-esp-container
    - https://github.com/dflemstr/rust-esp-container
    - https://github.com/jeikabu/rust-esp-container (disclaimer: mine)

Probably several more...

georgik, 2021-07-23:
There is container with v1.50 based on ivmarkov's repo with STD and Xtensa using pre-built toolchains which is using espressif/esp-idf Ubuntu 18 base image - built by GitHub Action: https://github.com/espressif/rust-esp32-example/blob/main/.github/workflows/publish-esp-idf-rust-image.yml
Dockerfile: https://github.com/espressif/rust-esp32-example/blob/main/Dockerfile

It's available also as part of GitHub Action https://github.com/marketplace/actions/espressif-iot-development-framework-esp-idf?version=rust-latest

The example Action is using the image: https://github.com/espressif/rust-esp32-example/actions/workflows/test-build.yml

You can pull it: `docker pull georgikrocks/esp-idf-rust`

Let's discuss this topic during next meeting #56

georgik, 2021-08-10:
Docker image moved to official Espressif DockerHub account: `docker pull espressif/idf-rust`

georgik, 2021-08-13:
Update: there are two images available right now

- CI image: `docker run -it espressif/idf-rust`
- GitHub action: https://github.com/marketplace/actions/espressif-iot-development-framework-esp-idf?version=rust-latest
- image with examples: `docker run -it espressif/idf-rust-examples`

Images are based on following Dockerfiles:
- espressif/idf-rust layers on top of idf base image: https://github.com/espressif/rust-esp32-example/blob/main/support/idf-rust/Dockerfile
- espressif/idf-rust-examples layers on top of idf-rust base image: https://github.com/espressif/rust-esp32-example/blob/main/Dockerfile

Images were built and pushed by GitHub runner:
- https://github.com/espressif/rust-esp32-example/blob/main/.github/workflows/publish-idf-rust-image.yml
- https://github.com/espressif/rust-esp32-example/blob/main/.github/workflows/publish-idf-rust-examples-image.yml

The dist files with Rust compiler for images mentioned above were built mostly on GitHub using:
- https://github.com/georgik/idf-rust-build
The build process is still in development since some arch like Windows x64 and macOS M1 are causing troubles.

Note: Flashing does not work from containers, it's necessary to copy the file from container and flash it from host machine. It's possible to copy file by command like: `docker cp ...`

With this information I'm closing the issue. Feel free to reopen the issue in case of problems with images.

closed 2021-08-13 by georgik