esp-rs/rust #42: Atomic operation caused IllegalInstruction on esp32s2

Neo-Zhixing, 2020-11-08:
I tried this code:

```rust
let a = AtomicUsize::new(1);
a.compare_and_swap(3, 3, Ordering::Release);
```

and this triggered a panic:
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40087540: c192280c 130c8000 a100e982
0x40087540: esprs::executor::Executor<T>::run at ??:?

Are atomic operations supported in this implementation?

Thank you!

I expected to see this happen: *explanation*

Instead, this happened: *explanation*

### Meta

`rustc --version --verbose`:
```
<version>
```

<details><summary>Backtrace</summary>
<p>

```
<backtrace>
```

</p>
</details>

MabezDev, 2020-12-11:
The esp32s2 uses an xtensa-lx7 processor, its likely the atomic implementation for xtensa-lx6 (what the esp32 uses) won't work. This is not a rust issue, but a llvm issue, I'd recommend asking over at https://github.com/espressif/llvm-project.

Neo-Zhixing, 2020-12-13:
Thanks for the hint!

closed 2020-12-13 by Neo-Zhixing