esp-rs/rust #46: New targets for esp32s2 and esp32s3

ivmarkov, 2021-04-05:
Given that for these MCUs Espressif provides separate toolchains, perhaps we need separate targets too. At the very least, the `linker` field in `TargetOptions` as well as the `cpu`  should be different.

Also: shall we create a target for esp32c3? I get it that it is not really 'xtensa', but it needs the same changes. Possibly based on some of the existing "riscv-" targets?

ivmarkov, 2021-04-05:
... and while we are at it, how about putting some extra info in the targets, like `unix`, vendor = `espressif` and env = `newlib`. Mentioned [here](https://github.com/MabezDev/rust-xtensa/pull/17#issuecomment-787905283).

For those who have no interest in using this with the ESP-IDF SDK, these would do no harm, as they won't be trying to build the STD crates anyway. However, for me this extra info is very beneficial, as most of my patches STD are checking for generic stuff, like `cfg(unix)` and `target_env = newlib`. A few also check for `target_os = none` and `target_vendor = espressif`.

MabezDev, 2021-04-15:
Hi @ivmarkov,

I've just updated the fork to 1.51, with a target def for the esp32s2, it seems esp32s3 support has not yet landed in llvm.

> Also: shall we create a target for esp32c3?

I'm not yet convinced this is necessary, and I think we should try and lean on upstream support where we can until we really need custom definitions.

As for the custom defs for libstd porting, I'm not apposed to them haven't really looked into the libstd porting effort enough yet.

ivmarkov, 2021-07-22:
Closing as the esp32s2 target is there, and the ESP-IDF-specific targets for Xtensa & RiscV are part of the libstd pull request.

closed 2021-07-22 by ivmarkov