esp-rs/rust #48: ICE on 1.51 when building esp32-hal's alloc example

kelnos, 2021-04-22:
Using esp32-hal https://github.com/esp-rs/esp32-hal/tree/94dc49fed351246ee57e9c3a4a28fe662b11165d and building with:
```
cargo xbuild --release --features='mem,rt,alloc' --example alloc
```
I get:
```
error: internal compiler error: compiler/rustc_mir/src/monomorphize/collector.rs:826:9: no MIR available for DefId(8:1581 ~ esp32_hal[a18f]::alloc::DEFAULT_HEAP)

thread 'rustc' panicked at 'Box<Any>', /home/brian/src/esp32-rust/rust-xtensa/library/std/src/panic.rs:59:5
stack backtrace:
   0:     0x778d6a92bf5c - <std::sys_common::backtrace::_print::DisplayBacktrace as core::fmt::Display>::fmt::h92f05c32d34e55b1
   1:     0x778d6a9e65ff - core::fmt::write::habc01af56514e184
   2:     0x778d6a92aae6 - std::io::Write::write_fmt::h786298d3493045df
   3:     0x778d6a94cb65 - std::panicking::default_hook::{{closure}}::h9f81f29c173dd1ba
   4:     0x778d6a94c733 - std::panicking::default_hook::h29961f4fed4671b5
   5:     0x778d6b3f35b4 - rustc_driver::report_ice::h62b7dd6e3311caf9
   6:     0x778d62e6b7f6 - <alloc::boxed::Box<F,A> as core::ops::function::Fn<Args>>::call::hdb2823b27ceaf625
                               at /home/brian/src/esp32-rust/rust-xtensa/library/alloc/src/boxed.rs:1535:9
   7:     0x778d62e69019 - proc_macro::bridge::client::<impl proc_macro::bridge::Bridge>::enter::{{closure}}::{{closure}}::he8834c1ed5f8ce0d
                               at /home/brian/src/esp32-rust/rust-xtensa/library/proc_macro/src/bridge/client.rs:320:21
   8:     0x778d6a94d2ca - std::panicking::rust_panic_with_hook::h8508dce83d0c4d0e
   9:     0x778d70becfbd - std::panicking::begin_panic::{{closure}}::he4d98f1dc0eef9ed
  10:     0x778d70bec4e6 - std::sys_common::backtrace::__rust_end_short_backtrace::hb6db84595d50ab6f
  11:     0x778d70bc6b5f - std::panicking::begin_panic::he1006c7b8d50deed
  12:     0x778d70bf848d - std::panic::panic_any::h730abc2446882085
  13:     0x778d70bfe9f5 - rustc_errors::HandlerInner::bug::h94fd4ed00a59c29a
  14:     0x778d70bfd130 - rustc_errors::Handler::bug::hfa492ecff8df8ec3
  15:     0x778d7064f17d - rustc_middle::util::bug::opt_span_bug_fmt::{{closure}}::h01a25b32ddfce489
  16:     0x778d70649c0b - rustc_middle::ty::context::tls::with_opt::{{closure}}::hffd33416f2fe257f
  17:     0x778d706496d2 - rustc_middle::ty::context::tls::with_opt::hcbf9ecbe823a5213
  18:     0x778d7064f0a3 - rustc_middle::util::bug::opt_span_bug_fmt::hdc326236b8dcf6d5
  19:     0x778d7064f015 - rustc_middle::util::bug::bug_fmt::h2c5e53dec68bdf1e
  20:     0x778d6f0aced1 - rustc_mir::monomorphize::collector::should_codegen_locally::h2ecb295070b6c359
  21:     0x778d6f0aeb58 - rustc_mir::monomorphize::collector::collect_miri::h2fdad49e50385172
  22:     0x778d6f0a9ee2 - rustc_mir::monomorphize::collector::collect_items_rec::h35a4e3f6b96ff7b2
  23:     0x778d6f3f24e1 - rustc_session::utils::<impl rustc_session::session::Session>::time::h639d283fc3e9aa39
  24:     0x778d6f0a8e2a - rustc_mir::monomorphize::collector::collect_crate_mono_items::h6749e3d27b803dd5
  25:     0x778d6f1de637 - rustc_mir::monomorphize::partitioning::collect_and_partition_mono_items::h2e9f893f2a32a691
  26:     0x778d6b7fe962 - rustc_middle::ty::query::<impl rustc_query_system::query::config::QueryAccessors<rustc_middle::ty::context::TyCtxt> for rustc_middle::ty::query::queries::collect_and_partition_mono_items>::compute::h187cc8aed29e59f3
  27:     0x778d6b7204be - rustc_middle::dep_graph::<impl rustc_query_system::dep_graph::DepKind for rustc_middle::dep_graph::dep_node::DepKind>::with_deps::hc22bd7ec540ef31e
  28:     0x778d6b89cbef - rustc_query_system::dep_graph::graph::DepGraph<K>::with_task_impl::hb201ace3ced09119
  29:     0x778d6b7bfdc6 - rustc_data_structures::stack::ensure_sufficient_stack::hfba4598167b79f32
  30:     0x778d6b83b65a - rustc_query_system::query::plumbing::force_query_with_job::h2f88a763e5fb37b8
  31:     0x778d6b816521 - rustc_query_system::query::plumbing::get_query_impl::h1a4b12822e21fec3
  32:     0x778d6b810729 - <rustc_codegen_llvm::LlvmCodegenBackend as rustc_codegen_ssa::traits::backend::CodegenBackend>::codegen_crate::haa0696e63f406593
  33:     0x778d6b61b89d - rustc_interface::passes::QueryContext::enter::h81c882d5ed9ca0aa
  34:     0x778d6b66c633 - rustc_interface::queries::Queries::ongoing_codegen::h5e1d85538a562eab
  35:     0x778d6b4186eb - rustc_interface::queries::<impl rustc_interface::interface::Compiler>::enter::hc8eb60c36b7135e8
  36:     0x778d6b3f77be - rustc_span::with_source_map::h4e873b7fc2994ddf
  37:     0x778d6b419b1b - rustc_interface::interface::create_compiler_and_run::h6b6b3cbc7bad9a01
  38:     0x778d6b3f8870 - rustc_span::with_session_globals::h23089b9f3c681cae
  39:     0x778d6b41b880 - std::sys_common::backtrace::__rust_begin_short_backtrace::h098db73ee66fea81
  40:     0x778d6b44448a - core::ops::function::FnOnce::call_once{{vtable.shim}}::he83ff58da95d651c
  41:     0x778d6a95bdf8 - std::sys::unix::thread::Thread::new::thread_start::h9e66dc0481aecb2f
  42:     0x778d6a679ea7 - start_thread
  43:     0x778d6a7b5def - clone
  44:                0x0 - <unknown>

note: the compiler unexpectedly panicked. this is a bug.

note: we would appreciate a bug report: https://github.com/rust-lang/rust/issues/new?labels=C-bug%2C+I-ICE%2C+T-compiler&template=ice.md

note: rustc 1.51.0-nightly (c4c02e837 2021-04-15) running on x86_64-unknown-linux-gnu

note: compiler flags: -C embed-bitcode=no -C debuginfo=2 -C incremental -C link-arg=-nostartfiles -C link-arg=-Wl,-Tlink.x --crate-type bin

note: some of the compiler flags provided by cargo are hidden

query stack during panic:
#0 [collect_and_partition_mono_items] collect_and_partition_mono_items
end of query stack
error: aborting due to previous error
```

MabezDev, 2021-04-22:
Reported upstream, https://github.com/rust-lang/rust/issues/84455

lehmanju, 2021-07-15:
Seems to be fixed in nightly https://github.com/rust-lang/rust/issues/84455#issuecomment-858615036

MabezDev, 2021-08-01:
The current 1.54 based esp branch has this fix included.

closed 2021-08-01 by MabezDev