esp-rs/rust #60: LLVM ERROR: Not supported instr: <MCInst 237 <MCOperand Reg:42>>

KerryRJ, 2021-07-30:
I am trying to build rust, on Windows 10 21h1, using both the default llvm-project and a custom llvm-project but am getting:-

>LLVM ERROR: Not supported instr: <MCInst 237 <MCOperand Reg:42>>
> 
> Did not run successfully: exit code: 101
"C:\\Dev\\Github\\rust\\build\\x86_64-pc-windows-msvc\\stage2\\bin\\rustc.exe" "--crate-name" "term" "--edition=2018" "library\\term\\src\\lib.rs" "--error-format=json" "--json=diagnostic-rendered-ansi" "--crate-type" "lib" "--emit=dep-info,metadata,link" "-C" "opt-level=3" "-C" "embed-bitcode=no" "-C" "debuginfo=0" "-C" "metadata=9ae1ecb98b4816ad" "-C" "extra-filename=-9ae1ecb98b4816ad" "--out-dir" "C:\\Dev\\Github\\rust\\build\\x86_64-pc-windows-msvc\\stage2-std\\xtensa-esp32-espidf\\release\\deps" "--target" "xtensa-esp32-espidf" "-C" "linker=xtensa-esp32-elf-cc" "-C" "incremental=C:\\Dev\\Github\\rust\\build\\x86_64-pc-windows-msvc\\stage2-std\\xtensa-esp32-espidf\\release\\incremental" "-L" "dependency=C:\\Dev\\Github\\rust\\build\\x86_64-pc-windows-msvc\\stage2-std\\xtensa-esp32-espidf\\release\\deps" "-L" "dependency=C:\\Dev\\Github\\rust\\build\\x86_64-pc-windows-msvc\\stage2-std\\release\\deps" "--extern" "core=C:\\Dev\\Github\\rust\\build\\x86_64-pc-windows-msvc\\stage2-std\\xtensa-esp32-espidf\\release\\deps\\libcore-d0f410522e28d1f8.rmeta" "--extern" "std=C:\\Dev\\Github\\rust\\build\\x86_64-pc-windows-msvc\\stage2-std\\xtensa-esp32-espidf\\release\\deps\\libstd-5d6c47f019a00a49.rlib" "-Zmacro-backtrace" "-Clink-args=-Wl,-rpath,$ORIGIN/../lib" "-Zsave-analysis" "-Cprefer-dynamic" "-Cllvm-args=-import-instr-limit=10" "-Cembed-bitcode=yes" "-Z" "binary-dep-depinfo" "-Wrust_2018_idioms" "-Wunused_lifetimes" "-Dwarnings" "--sysroot" "C:\\Dev\\Github\\rust\\build\\x86_64-pc-windows-msvc\\stage2" "-Z" "force-unstable-if-unmarked"
>
> error: could not compile `term`
>
>To learn more, run the command again with --verbose.
warning: build failed, waiting for other jobs to finish...
error: build failed
command did not execute successfully: "\\\\?\\C:\\Dev\\Github\\rust\\build\\x86_64-pc-windows-msvc\\stage0\\bin\\cargo.exe" "build" "--target" "xtensa-esp32-espidf" "-Zbinary-dep-depinfo" "-j" "6" "--release" "--features" "panic-unwind backtrace" "--manifest-path" "C:\\Dev\\Github\\rust\\library/test/Cargo.toml" "--message-format" "json-render-diagnostics"
expected success, got: exit code: 101
Traceback (most recent call last):
  File "C:\Dev\Github\rust\x.py", line 27, in <module>
    bootstrap.main()
  File "C:\Dev\Github\rust\src\bootstrap\bootstrap.py", line 1191, in main
    bootstrap(help_triggered)
  File "C:\Dev\Github\rust\src\bootstrap\bootstrap.py", line 1177, in bootstrap
    run(args, env=env, verbose=build.verbose)
  File "C:\Dev\Github\rust\src\bootstrap\bootstrap.py", line 153, in run
    raise RuntimeError(err)
RuntimeError: failed to run: C:\Dev\Github\rust\build\bootstrap\debug\bootstrap build --stage 2 --verbose`

I have migrated the changes to the rust repo master branch and get:-

>error: fixup value out of range
>
>LLVM ERROR: Cannot select: 0x1e75141f3e8: f32 = fma 0x1e75141f110, 0x1e75141f1e0, 0x1e75141f380
  0x1e75141f110: f32 = bitcast 0x1e75141ef08
    0x1e75141ef08: i32,ch = CopyFromReg 0x1e73c8d7da8, Register:i32 %0
      0x1e75141eea0: i32 = Register %0
  0x1e75141f1e0: f32 = bitcast 0x1e75141f0a8
    0x1e75141f0a8: i32,ch = CopyFromReg 0x1e73c8d7da8, Register:i32 %2
      0x1e75141f040: i32 = Register %2
  0x1e75141f380: f32 = fma 0x1e75141f318, 0x1e75141f178, 0x1e75141f178
    0x1e75141f318: f32 = fneg 0x1e75141f110
      0x1e75141f110: f32 = bitcast 0x1e75141ef08
        0x1e75141ef08: i32,ch = CopyFromReg 0x1e73c8d7da8, Register:i32 %0
          0x1e75141eea0: i32 = Register %0
    0x1e75141f178: f32 = bitcast 0x1e75141efd8
      0x1e75141efd8: i32,ch = CopyFromReg 0x1e73c8d7da8, Register:i32 %1
        0x1e75141ef70: i32 = Register %1
    0x1e75141f178: f32 = bitcast 0x1e75141efd8
      0x1e75141efd8: i32,ch = CopyFromReg 0x1e73c8d7da8, Register:i32 %1
        0x1e75141ef70: i32 = Register %1
In function: _ZN3std3f3221_$LT$impl$u20$f32$GT$4lerp17hd25dd7f6def92bafE
error: could not compile `std`
>
>To learn more, run the command again with --verbose.
command did not execute successfully: "\\\\?\\C:\\Dev\\Github\\rust\\build\\x86_64-pc-windows-msvc\\stage0\\bin\\cargo.exe" "build" "--target" "xtensa-esp32-espidf" "-Zbinary-dep-depinfo" "-j" "6" "--release" "--features" "panic-unwind backtrace compiler-builtins-c" "--manifest-path" "C:\\Dev\\Github\\rust\\library/test/Cargo.toml" "--message-format" "json-render-diagnostics"
expected success, got: exit code: 101
failed to run: C:\Dev\Github\rust\build\bootstrap\debug\bootstrap build --stage 2

Anyone point this newbie in a direction to troubleshoot this?

ivmarkov, 2021-07-30:
Hmmm. So you are trying to build the Rust **compiler**, right?
Are you following the instructions for building it exactly?
This looks very suspicious to me:
`"--target" "xtensa-esp32-espidf"`

Any `.config/cargo.toml` file lurking somewhere in a parent directory of your rust compiler directory?

KerryRJ, 2021-08-02:
Cleaned workstation, retried instructions instructions, finally got the compiler building.

closed 2021-08-02 by KerryRJ