esp-rs/rust #62: esp 1.54 breaks Mac M1 builds

MabezDev, 2021-08-01:
1.54 Rust introduces a new compiler builtins version which is broken on M1 Mac's with LLVM 11.

`cargo:warning=clang: error: no such file or directory: 'rust-xtensa-dev/src/llvm-project/compiler-rt/lib/builtins/aarch64/lse.S'`

I believe this will be fixed once the llvm fork is updated to LLVM 12, as the file [only exists in LLVM 12](https://github.com/llvm/llvm-project/tree/release/12.x/compiler-rt/lib/builtins/aarch64).

georgik, 2021-08-05:
Confirmed. I was able to reproduce the same bug using self-hosted GitHub runner. FYI @andreisfr

Build of 1.53 does not have the issue. Instructions for installing binary artifacts: https://github.com/espressif/rust-esp32-example/blob/main/docs/rust-on-xtensa-installation-aarch64-apple-darwin.md

georgik, 2021-08-09:
macOS M1 Rust 1.54.0 artifact is ready - it was built using LLVM 12 branch, which is under review now:
https://github.com/espressif/rust-esp32-example/blob/feature/rust-1.54.0/docs/rust-on-xtensa-installation-aarch64-apple-darwin.md

georgik, 2021-08-18:
Public build will be possible after sync of espressif/llvm-project with LLV 12 support.
Meanwhile it's possible to use binary artifact mentioned above.

MabezDev, 2021-08-20:
It's now public - I've updated the esp branch to use said updated fork.

closed 2021-08-20 by MabezDev