esp-rs/rust #64: Can't use inline assembly

NathanMcMillan54, 2021-08-03:
When I try to compile anything that uses an assembly macro on any xtensa target, I get this error:

```command line
error[E0472]: inline assembly is unsupported on this target
```
This is my ``.cargo/config.toml``:

```toml
[target.xtensa-esp32-none-elf]
rustflags = [
    "-C", "link-arg=-nostartfiles",
]
```

And I compiled with:
```command line
cargo xbuild --target xtensa-esp32-none-elf
```
Did I compile the compiler wrong? Or did I compile the project wrong?

MabezDev, 2021-08-03:
Typed Inline asm has not been ported to the xtensa arch yet, see #32. You can however use the `llvm_asm!` macro.

NathanMcMillan54, 2021-08-04:
But when I try to compile [xtensa-lx-rt](https://github.com/esp-rs/xtensa-lx-rt) I get this error because of the ``global_asm!`` macro. How does that get compiled?

MabezDev, 2021-08-04:
I see, it seems `global_asm!` has been ported to the `asm!` format, without providing a `global_llvm_asm!` alternative. I'll see what I can do to work around this.

MabezDev, 2021-08-13:
Hi @NathanMcMillan54, sorry about the trouble - I have added support for the new `asm!` format and merged it into the esp branch. New releases of xtensa-lx, xtensa-lx-rt have been released too with support for this assembly syntax.

Please report back if you run into any issues - hope this helps :)

closed 2021-08-13 by MabezDev

NathanMcMillan54, 2021-08-13:
Oh thanks! I'll see if it worked

NathanMcMillan54, 2021-08-19:
This isn't related to this issue, when I recompiled Xtensa Rust and try to compile anything I get this error:
```command line
error: linking with `xtensa-esp32-elf-gcc` failed: exit status: 1
  |
  = note: "xtensa-esp32-elf-gcc" .... a ton of file paths....

    ....

  = note: xtensa-esp32-elf-gcc: fatal error: -fuse-linker-plugin, but liblto_plugin.so not found
          compilation terminated.

```

Is ``liblto_plugin.so`` a file I need to install?

MabezDev, 2021-08-19:
Seems like you have an old version on xtensa gcc? Try with the latest from here: https://github.com/espressif/crosstool-NG/releases.

NathanMcMillan54, 2021-08-19:
Thanks! I'll try that