esp-rs/rust #72: Support for installation of custom toolchain with Xtensa support via rustup

georgik, 2021-08-12:
Observed state:
- installation is manual and requires several steps: https://github.com/espressif/rust-esp32-example/blob/main/docs/rust-on-xtensa.md#quick-start

Desired state:
- installation is possible via rustup command

georgik, 2021-08-12:
Rustup supports custom toolchain when following variable is set RUSTUP_DIST_SERVER.

Question: What is the layout and how we can map our dist files to the layout and how to create index file with metadata?

igrr, 2021-08-12:
Linking https://internals.rust-lang.org/t/future-updates-to-the-rustup-distribution-format/4196 which seems to answer the 2nd question. Not sure if anything important has changed about the layout between 2017 and now.

jessebraham, 2021-08-19:
I'm posting a bunch of information here, for lack of a better place. I may compile a full document at some point better explaining all this.

- - -

### Directory Structure

The root directory in the below tree is `RUSTUP_DIST_SERVER`.  
The `rustup/` directory within the root is `RUSTUP_UPDATE_ROOT`.

```text
$CHANNEL - release channel, one of {stable,beta,nightly}
$DATE    - release date in the format YYYY-MM-DD
$ARCH    - architecture triple, eg.) x86_64-unknown-linux-gnu

.
├── dist
│   ├── $DATE
│   │   ├── $PACKAGE-$CHANNEL-$ARCH.tar.gz
│   │   ├── $PACKAGE-$CHANNEL-$ARCH.tar.gz.sha256
│   │   ├── $PACKAGE-$CHANNEL-$ARCH.tar.xz
│   │   ├── $PACKAGE-$CHANNEL-$ARCH.tar.xz.sha256
│   │   └── ...
│   ├── channel-rust-$CHANNEL.toml
│   └── channel-rust-$CHANNEL.toml.sha256
└── rustup
    ├── dist
    │   └── $ARCH
    │       └── rustup-init
    └── release-stable.toml
```

### Manifest Format

```toml
manifest-version = "2"
date = "YYYY-MM-DD"

# Each package has configuration as follows. Many targets
# can be specified for each package.
[pkg.cargo]
git_commit_hash = ""
version = ""

[pkg.cargo.target.x86_64-unknown-linux-gnu]
available = true
hash = ""
url = ""
xz_hash = ""
xz_url = ""

[pkg.cargo.target.aarch64-unknown-linux-gnu]
available = false

# `installer-msi`, `installer-pkg` and `source-code` are
# all valid artifact types.
[[artifacts.source-code.target."*"]]
hash-sha256 = ""
url = ""

# Allows packages to be renamed; not sure why this is done.
[renames.rustfmt]
to = "rustfmt-preview"

# Specifies which packages are included for each profile.
[profiles]
complete = []
default = []
minimal = []
```

georgik, 2021-09-16:
Closing: Rustup limitation does not allow to deploy custom toolchain from custom location.

closed 2021-09-16 by georgik