esp-rs/rust #79: Build the core crate in the CI

lexxvir, 2021-09-12:
Hi there,

It would be nice, if the CI that builds Rust compiler for Xtensa also build and distribute the core crate for Xtensa arch.

I know that in the embedded world it's usual case to build the core crate with specific optimizations (debug vs release) that required in particular case. But in my daily work I never touch or rebuild the core crate.

Thanks in advance!

georgik, 2022-01-12:
Thank you @lexxvir for the suggestion.
We already included some tools in binary form in 1.57.0.2 build: `--tools=clippy,cargo,rustfmt`
Example of build workflow: https://github.com/esp-rs/rust-build/blob/main/.github/actions/build/build-tools/action.yaml#L14

Any about which flag is required to build the binary core and include it in dist artifact?
Configuration script in Rust source code dir is:`src/bootstrap/configure.py`

The issue scheduled for v1.58 build.

lexxvir, 2022-01-12:
Hi @georgik !

Sorry, I don't certainly know what should be done. I think we need the same "settings" as for thumb* targets.
After brief searching I found two places in the repo that need to tweak:
1. The [Dockerfile](https://github.com/rust-lang/rust/blob/7baa7afd0fc8061c5b815a7aab9b40d1cd406ce8/src/ci/docker/host-x86_64/dist-various-1/Dockerfile)
2. And maybe the [build-manifest](https://github.com/rust-lang/rust/blob/db1fb85cff63ad5fffe435e17128f99f9e1d970c/src/tools/build-manifest/src/main.rs).

MabezDev, 2022-11-03:
We heavily rely on the build-std feature for both core and std builds so I don't think we'll look to do this now, thanks for the suggestion though!

closed 2022-11-03 by MabezDev