esp-rs/rust #81: UdpSocket::bind cannot be linked

mladedav, 2021-09-28:
I tried to use the std library for rust on esp32 but I am running to problems when compiling code that should bind a udp socket. I get an error about undefined reference to bind. If I comment out the networking work (which leaves me with a mostly empty program), the code compiles.

I tried both with TcpListener and UdpSocket and both fail on the linking phase.

I am sorry if I am making something wrong because I can see that https://github.com/ivmarkov/rust-esp32-std-hello makes use of `bind`, but when I use that feature there, the linker fails with the same error.

I tried this code:

```rust
    let socket = UdpSocket::bind("127.0.0.1:0").unwrap();
```

I expected to see this happen: *explanation*

Instead, this happened: 

```
Running ldproxy
          Error: Linker C:\Users\name\.platformio\packages\toolchain-xtensa32\bin\xtensa-esp32-elf-gcc.EXE failed: exit code: 1
          STDERR OUTPUT:
          c:/users/name/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:\Users\name\project\target\xtensa-esp32-espidf\debug\deps\libstd-eb631c2455bb8cbc.rlib(std-eb631c2455bb8cbc.std.66ojinsg-cgu.12.rcgu.o):(.literal._ZN3std10sys_common3net9UdpSocket4bind17hb79019ec8f516013E+0x18): undefined reference to `bind'
          collect2.exe: error: ld returned 1 exit status
```

### Meta

`rustc --version --verbose`:
```
rustc 1.54.0-dev
binary: rustc
commit-hash: unknown
commit-date: unknown
host: x86_64-pc-windows-msvc
release: 1.54.0-dev
LLVM version: 11.0.0
```
OS: Windows 10

I am using the esp toolchain, installation according to https://github.com/espressif/rust-esp32-example/blob/main/docs/rust-on-xtensa-installation-x86_64-pc-windows-msvc.md
I'm targeting `xtensa-esp32-espidf`

MabezDev, 2021-09-28:
Please update to 1.55, which includes the libc fixes few `bind` and a few others :).

mladedav, 2021-09-28:
@MabezDev I will, thank you. Just to check, I probably have to build that from source?

I downloaded this one from https://dl.espressif.com/dl/idf-rust/dist/x86_64-pc-windows-msvc/rust-1.54.0-dev-x86_64-pc-windows-msvc.zip but changing the version to 1.55 (and dropping the dev) results in 404.

jessebraham, 2021-09-28:
You can find pre-built artifacts here:
https://github.com/esp-rs/rust-build/

mladedav, 2021-09-29:
Upgrading to 1.55 solved the issue.

Thank you both for your assistance.

closed 2021-09-29 by mladedav