esp-rs/rust #93: error: linking with `xtensa-lx106-elf-gcc` failed: exit status: 1

jessebraham, 2021-11-18:
Using `rustc-1.56.0.1` installed from the _rust-build_ repository. Building for the ESP8266 in `debug` mode seems to result in the following error:

```
error: linking with `xtensa-lx106-elf-gcc` failed: exit status: 1
  |
  = note: "xtensa-lx106-elf-gcc" "/Users/jesse/Work/esp8266-hal/target/xtensa-esp8266-none-elf/debug/examples/blinky-e0818fc35520ce14.panic_halt-43f7212d7c513c73.panic_halt.f3fb034b-cgu.0.rcgu.o.rcgu.o" "-Wl,--as-needed" "-L" "/Users/jesse/Work/esp8266-hal/target/xtensa-esp8266-none-elf/debug/deps" "-L" "/Users/jesse/Work/esp8266-hal/target/debug/deps" "-L" "/Users/jesse/Work/esp8266-hal/target/xtensa-esp8266-none-elf/debug/build/esp8266-hal-2d03f26e6b6aa8d6/out" "-L" "/Users/jesse/Work/esp8266-hal/target/xtensa-esp8266-none-elf/debug/build/xtensa-lx-9a48fc8303ca0ffa/out" "-L" "/Users/jesse/Work/esp8266-hal/target/xtensa-esp8266-none-elf/debug/build/xtensa-lx-rt-02f5e2b7b68604e9/out" "-L" "/Users/jesse/.rustup/toolchains/esp/lib/rustlib/xtensa-esp8266-none-elf/lib" "-Wl,--start-group" "-Wl,--end-group" "-Wl,-Bstatic" "/Users/jesse/Work/esp8266-hal/target/xtensa-esp8266-none-elf/debug/deps/libcompiler_builtins-8b73f41dc98a2660.rlib" "-Wl,-Bdynamic" "-Wl,--eh-frame-hdr" "-Wl,-znoexecstack" "-L" "/Users/jesse/.rustup/toolchains/esp/lib/rustlib/xtensa-esp8266-none-elf/lib" "-o" "/Users/jesse/Work/esp8266-hal/target/xtensa-esp8266-none-elf/debug/examples/blinky-e0818fc35520ce14" "-Wl,--gc-sections" "-no-pie" "-nodefaultlibs" "-nostartfiles" "-Wl,-Tlink.x"
  = note: /Users/jesse/.xtensa/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/8.4.0/../../../../xtensa-lx106-elf/bin/ld: /Users/jesse/Work/esp8266-hal/target/xtensa-esp8266-none-elf/debug/deps/libcompiler_builtins-8b73f41dc98a2660.rlib(compiler_builtins-8b73f41dc98a2660.compiler_builtins.77efb3e9-cgu.3.rcgu.o):(.literal._ZN17compiler_builtins3int19specialized_div_rem23u32_normalization_shift17h93377cdcc12e9cdcE+0x8): undefined reference to `core::panicking::panic'
          collect2: error: ld returned 1 exit status

  = help: some `extern` functions couldn't be found; some native libraries may need to be installed or have their path specified
  = note: use the `-l` flag to specify native libraries to link
  = note: use the `cargo:rustc-link-lib` directive to specify the native libraries to link with Cargo (see https://doc.rust-lang.org/cargo/reference/build-scripts.html#cargorustc-link-libkindname)

error: could not compile `esp8266-hal` due to previous error
```

Curiously, building in `release` seems to resolve the issue. This can be reproduced by executing the following:

```shell
$ git clone https://github.com/esp-rs/esp8266-hal.git
$ cd esp8266-hal/
$ cargo build --example=blinky           # fails!
$ cargo build --example=blinky --release # succeeds!
```

The `Cargo.toml` manifest for this repository contains the following configuration for build profiles, not sure if `lto` is the culprit or if it's something else:

```toml
[profile.dev]
lto = true
opt-level = 1

[profile.release]
lto = true
```

jessebraham, 2021-11-18:
I've disabled `lto` for the `dev` profile and this seems to have resolved the linker error.

MabezDev, 2021-11-24:
I think this is related to https://github.com/rust-lang/compiler-builtins/issues/347. Doesn't look like there is a solution yet, just some work arounds & theories.

MabezDev, 2022-03-15:
Just following up on this, seems although this hasn't been fixed yet its been tracked down to https://github.com/rust-lang/cargo/issues/10118#issuecomment-1006000210. This is why it works in release, because there are no overflow checks etc.

closed 2022-07-15 by MabezDev