esp-rs/rust #100: Link error when using some `std::fs` functions

Eroc33, 2022-01-02:
When using `std::fs::File.set_len` anywhere in the project a link error occurs referencing `ftruncate`:

Relevant excerpt
```
/home/euan/esp-idf-ftruncate-link-bug-repro/.embuild/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /home/euan/esp-idf-ftruncate-link-bug-repro/target/xtensa-esp32-espidf/debug/deps/libstd-d1f4638178e555e4.rlib(std-d1f4638178e555e4.std.43feb998-cgu.2.rcgu.o):(.literal._ZN3std3sys4unix5cvt_r17h6b170249ec86b082E+0x0): undefined reference to `ftruncate'
```

<details>
<summary>Full build output</summary>
<pre>
euan@Euan-Desktop:~/esp-idf-ftruncate-link-bug-repro$ cargo build
    Blocking waiting for file lock on build directory
   Compiling bindgen v0.59.2
   Compiling embuild v0.27.1
   Compiling esp-idf-sys v0.28.3
   Compiling esp-idf-ftruncate-link-bug-repro v0.1.0 (/home/euan/esp-idf-ftruncate-link-bug-repro)
error: linking with `ldproxy` failed: exit status: 1
  |
  = note: "ldproxy" "/home/euan/esp-idf-ftruncate-link-bug-repro/target/xtensa-esp32-espidf/debug/deps/esp_idf_ftruncate_link_bug_repro-f438ed8d59ada69c.1iq3gu07s860atys.rcgu.o" "/home/euan/esp-idf-ftruncate-link-bug-repro/target/xtensa-esp32-espidf/debug/deps/esp_idf_ftruncate_link_bug_repro-f438ed8d59ada69c.1xhrgistnoo3vrvu.rcgu.o" "/home/euan/esp-idf-ftruncate-link-bug-repro/target/xtensa-esp32-espidf/debug/deps/esp_idf_ftruncate_link_bug_repro-f438ed8d59ada69c.3p5lv2lqfhd1l9sn.rcgu.o" "/home/euan/esp-idf-ftruncate-link-bug-repro/target/xtensa-esp32-espidf/debug/deps/esp_idf_ftruncate_link_bug_repro-f438ed8d59ada69c.3q5y8i7sp4xzdsye.rcgu.o" "/home/euan/esp-idf-ftruncate-link-bug-repro/target/xtensa-esp32-espidf/debug/deps/esp_idf_ftruncate_link_bug_repro-f438ed8d59ada69c.4qyt2qwp1t65e50g.rcgu.o" "/home/euan/esp-idf-ftruncate-link-bug-repro/target/xtensa-esp32-espidf/debug/deps/esp_idf_ftruncate_link_bug_repro-f438ed8d59ada69c.28w0hodb6thz5p10.rcgu.o" "-Wl,--as-needed" "-L" "/home/euan/esp-idf-ftruncate-link-bug-repro/target/xtensa-esp32-espidf/debug/deps" "-L" "/home/euan/esp-idf-ftruncate-link-bug-repro/target/debug/deps" "-L" "/home/euan/.rustup/toolchains/esp/lib/rustlib/xtensa-esp32-espidf/lib" "-Wl,-Bstatic" "/home/euan/esp-idf-ftruncate-link-bug-repro/target/xtensa-esp32-espidf/debug/deps/libesp_idf_sys-c71515d603b41be1.rlib" "/home/euan/esp-idf-ftruncate-link-bug-repro/target/xtensa-esp32-espidf/debug/deps/libmutex_trait-df9aa576e9040a2f.rlib" "/home/euan/esp-idf-ftruncate-link-bug-repro/target/xtensa-esp32-espidf/debug/deps/libembedded_svc-2899069fe8327a24.rlib" "/home/euan/esp-idf-ftruncate-link-bug-repro/target/xtensa-esp32-espidf/debug/deps/liblog-bcbaccb88823ec70.rlib" "/home/euan/esp-idf-ftruncate-link-bug-repro/target/xtensa-esp32-espidf/debug/deps/libcfg_if-935bd39d9f6427e5.rlib" "/home/euan/esp-idf-ftruncate-link-bug-repro/target/xtensa-esp32-espidf/debug/deps/libanyhow-c3305a373346a996.rlib" "/home/euan/esp-idf-ftruncate-link-bug-repro/target/xtensa-esp32-espidf/debug/deps/libenumset-7515a396e404b77d.rlib" "-Wl,--start-group" "/home/euan/esp-idf-ftruncate-link-bug-repro/target/xtensa-esp32-espidf/debug/deps/libstd-d1f4638178e555e4.rlib" "/home/euan/esp-idf-ftruncate-link-bug-repro/target/xtensa-esp32-espidf/debug/deps/libpanic_abort-45242a1e085ac8c1.rlib" "/home/euan/esp-idf-ftruncate-link-bug-repro/target/xtensa-esp32-espidf/debug/deps/libstd_detect-0d85fdf205992f33.rlib" "/home/euan/esp-idf-ftruncate-link-bug-repro/target/xtensa-esp32-espidf/debug/deps/librustc_demangle-09912fe574e03e88.rlib" "/home/euan/esp-idf-ftruncate-link-bug-repro/target/xtensa-esp32-espidf/debug/deps/libhashbrown-b84fbcd507543c60.rlib" "/home/euan/esp-idf-ftruncate-link-bug-repro/target/xtensa-esp32-espidf/debug/deps/librustc_std_workspace_alloc-6f53877691d2c88b.rlib" "/home/euan/esp-idf-ftruncate-link-bug-repro/target/xtensa-esp32-espidf/debug/deps/libunwind-75acfd38a7bfff04.rlib" "/home/euan/esp-idf-ftruncate-link-bug-repro/target/xtensa-esp32-espidf/debug/deps/libcfg_if-16b2675bf0230332.rlib" "/home/euan/esp-idf-ftruncate-link-bug-repro/target/xtensa-esp32-espidf/debug/deps/liblibc-a626876b7c2b7226.rlib" "/home/euan/esp-idf-ftruncate-link-bug-repro/target/xtensa-esp32-espidf/debug/deps/liballoc-5a1058ec08d59625.rlib" "/home/euan/esp-idf-ftruncate-link-bug-repro/target/xtensa-esp32-espidf/debug/deps/librustc_std_workspace_core-17c3abae709388e4.rlib" "/home/euan/esp-idf-ftruncate-link-bug-repro/target/xtensa-esp32-espidf/debug/deps/libcore-8f5a7b792e1e0797.rlib" "-Wl,--end-group" "/home/euan/esp-idf-ftruncate-link-bug-repro/target/xtensa-esp32-espidf/debug/deps/libcompiler_builtins-20ef2192007b875d.rlib" "-Wl,-Bdynamic" "-Wl,--eh-frame-hdr" "-Wl,-znoexecstack" "-L" "/home/euan/.rustup/toolchains/esp/lib/rustlib/xtensa-esp32-espidf/lib" "-o" "/home/euan/esp-idf-ftruncate-link-bug-repro/target/xtensa-esp32-espidf/debug/deps/esp_idf_ftruncate_link_bug_repro-f438ed8d59ada69c" "-Wl,--gc-sections" "-no-pie" "-nodefaultlibs" "--ldproxy-linker" "/home/euan/esp-idf-ftruncate-link-bug-repro/.embuild/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc" "--ldproxy-dedup-libs" "-L/home/euan/esp-idf-ftruncate-link-bug-repro/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-722ca5ac09ebffa5/out/esp-idf" "-L/home/euan/esp-idf-ftruncate-link-bug-repro/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-722ca5ac09ebffa5/out/esp-idf/.pio/build/debug" "-L/home/euan/esp-idf-ftruncate-link-bug-repro/.embuild/platformio/packages/framework-espidf/components/esp_wifi/lib/esp32" "-L/home/euan/esp-idf-ftruncate-link-bug-repro/.embuild/platformio/packages/framework-espidf/components/esp_rom/esp32/ld" "-L/home/euan/esp-idf-ftruncate-link-bug-repro/.embuild/platformio/packages/framework-espidf/components/xtensa/esp32" "-L/home/euan/esp-idf-ftruncate-link-bug-repro/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-722ca5ac09ebffa5/out/esp-idf/.pio/build/debug/esp-idf/esp32" "-L/home/euan/esp-idf-ftruncate-link-bug-repro/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-722ca5ac09ebffa5/out/esp-idf/.pio/build/debug/esp-idf/esp32/ld" "-L/home/euan/esp-idf-ftruncate-link-bug-repro/.embuild/platformio/packages/framework-espidf/components/esp32/ld" "-Wl,--start-group" "/home/euan/esp-idf-ftruncate-link-bug-repro/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-722ca5ac09ebffa5/out/esp-idf/.pio/build/debug/esp-idf/esp_pm/libesp_pm.a" "/home/euan/esp-idf-ftruncate-link-bug-repro/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-722ca5ac09ebffa5/out/esp-idf/.pio/build/debug/esp-idf/asio/libasio.a" "/home/euan/esp-idf-ftruncate-link-bug-repro/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-722ca5ac09ebffa5/out/esp-idf/.pio/build/debug/esp-idf/cbor/libcbor.a" "/home/euan/esp-idf-ftruncate-link-bug-repro/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-722ca5ac09ebffa5/out/esp-idf/.pio/build/debug/esp-idf/unity/libunity.a" "/home/euan/esp-idf-ftruncate-link-bug-repro/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-722ca5ac09ebffa5/out/esp-idf/.pio/build/debug/esp-idf/cmock/libcmock.a" "/home/euan/esp-idf-ftruncate-link-bug-repro/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-722ca5ac09ebffa5/out/esp-idf/.pio/build/debug/esp-idf/coap/libcoap.a" "/home/euan/esp-idf-ftruncate-link-bug-repro/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-722ca5ac09ebffa5/out/esp-idf/.pio/build/debug/esp-idf/console/libconsole.a" "/home/euan/esp-idf-ftruncate-link-bug-repro/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-722ca5ac09ebffa5/out/esp-idf/.pio/build/debug/esp-idf/esp_adc_cal/libesp_adc_cal.a" "/home/euan/esp-idf-ftruncate-link-bug-repro/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-722ca5ac09ebffa5/out/esp-idf/.pio/build/debug/esp-idf/esp_hid/libesp_hid.a" "/home/euan/esp-idf-ftruncate-link-bug-repro/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-722ca5ac09ebffa5/out/esp-idf/.pio/build/debug/esp-idf/protobuf-c/libprotobuf-c.a" "/home/euan/esp-idf-ftruncate-link-bug-repro/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-722ca5ac09ebffa5/out/esp-idf/.pio/build/debug/esp-idf/protocomm/libprotocomm.a" "/home/euan/esp-idf-ftruncate-link-bug-repro/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-722ca5ac09ebffa5/out/esp-idf/.pio/build/debug/esp-idf/mdns/libmdns.a" "/home/euan/esp-idf-ftruncate-link-bug-repro/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-722ca5ac09ebffa5/out/esp-idf/.pio/build/debug/esp-idf/esp_local_ctrl/libesp_local_ctrl.a" "/home/euan/esp-idf-ftruncate-link-bug-repro/t [... cut ...]
  = note: use the `cargo:rustc-link-lib` directive to specify the native libraries to link with Cargo (see https://doc.rust-lang.org/cargo/reference/build-scripts.html#cargorustc-link-libkindname)

error: could not compile `esp-idf-lstat-link-bug-repro` due to previous error
</pre>
</details>

Repro project: https://github.com/Eroc33/esp-idf-lstat-link-bug-repro

As far as I can tell `ftruncate` and `lstat` simply don't exist in the esp-idf, so would need to be implemented there, or the libstd code that calls them would have to be implemented in terms of some other idf functions

MabezDev, 2022-03-03:
Thanks for the report @Eroc33, it's much appreciated! Sorry for the delay in getting to this.

The `lstat` fix is simple, as will be fixed in esp-idf in future releases, for now I have also patched it into `esp-idf-sys`, see https://github.com/esp-rs/esp-idf-sys/pull/75.

`ftruncate` is a bit more intricate, and will take a bit more time to get a patch prepared.

MabezDev, 2022-03-14:
Hi @Eroc33, `ftruncate` is now implemented in esp-idf (backport to v4.4 will be available soon) which should solve the issues you've encountered. Thanks for your patience!

closed 2022-03-14 by MabezDev