esp-rs/rust #117: Unable to list files in a SPIFFS partition

bugadani, 2022-04-22:
Given is the following test code (with some details omitted):

```rust
fn init_partition(path: &CStr, label: &CStr, max_files: usize) {
    let storage_conf = esp_vfs_spiffs_conf_t {
        base_path: path.as_ptr(),
        partition_label: label.as_ptr(),
        max_files,
        format_if_mount_failed: true,
    };

    if let Err(err) = EspErr::from(unsafe { esp_vfs_spiffs_register(&storage_conf) }) {
        match err {
            EspErr::Fail => log::error!("Failed to mount or format filesystem"),
            EspErr::NotFound => log::error!("Failed to find SPIFFS partition: {label}"),
            err => log::error!("Failed to initialize SPIFFS: {err}"),
        }
    }
}

fn test() {
    let path = std::path::Path::new("/storage/foobar");

    if let Err(err) = std::fs::File::create(path) {
        println!("Failed to create file: {err}");
        return;
    }

    match std::fs::read_dir("/storage") {
        Ok(dir) => {
            for entry in dir {
                println!("{entry:?}");
            }
        }
        Err(err) => println!("Failed to read root: {err}"),
    }

    if let Err(err) = std::fs::remove_file(path) {
        println!("Failed to remove file: {err}");
    }
}

#[no_mangle]
pub extern "C" fn app_main() {
    init_partition(c_str!("/storage"), c_str!("storage"), 3);

    test();
}
```

I expect the following output:

```
Ok(DirEntry("/storage/foobar"))
```

Instead, the output I get is the following:

```
Ok(DirEntry("/storage/"))
Err(Os { code: 5, kind: Uncategorized, message: "I/O error" })
```

The OS error is unrelated (SPIFFS seems to indicate the end of folder iteration by returning a value that gets turned into EIO), but note that the path in the DirEntry has an empty filename.

This issue is present on (table will be expanded as I have more information):

rustc verion    | esp-idf 4.4.1 | 
--------------- | -------------- |
Rustc 1.60.0.0 | x                  |
Rustc 1.59.0.1 | x                 |

MabezDev, 2022-05-30:
Hey @bugadani, thanks for your patience! I think I've tracked this down to another mismatch of structs on the C side and rust side.

The files are created, and when you step into the `ReadDir` Iterator, if you look carefully the entries are being populated; with one problem. A zeroed field not included in the C side happens to overlap with the start of the name field, meaning the name on the Rust side starts with a null pointer... hence why only the root directory is printed!

I need to track down whether this is an issue with rust-lang/libc or the newlib component of esp-idf.

MabezDev, 2022-05-30:
This is an esp-idf issue, specifically this [definition](https://github.com/espressif/esp-idf/blob/73be74fd71aeec03a972cfe0a9b2f19edc8ea3fa/components/newlib/platform_include/sys/dirent.h#L45).

`d_ino` should be type `ino_t`, but its defined as an `int`. `sizeof(int) == 4, sizeof(ino_t) == 2` on the esp-idf platform, hence this failure. I'll submit a fix asap :).

MabezDev, 2022-07-04:
Thanks for the report @bugadani. This is now fixed & backported to V4.4, so I'm closing this issue :). You may need to remove `.embuild/` and rebuild to pull in the fixes from the v4.4 branch.