esp-rs/rust #132: `std` build fails with newest rust nightly

N3xed, 2022-08-02:
The latest working nightly was the previous one, so `2022-08-01`. This also caused `esp-idf-sys` CI to fail (only the one that's using current nightly, the other three use esp-rs/rust).

<details>
<summary>Build output</summary>

**Errors**
```txt
error[E0425]: cannot find value `UTIME_OMIT` in crate `libc`
   --> C:\Users\n3xed\.rustup\toolchains\nightly-x86_64-pc-windows-msvc\lib\rustlib\src\rust\library\std\src\sys\unix\fs.rs:550:63
    |
550 |         let omit = libc::timespec { tv_sec: 0, tv_nsec: libc::UTIME_OMIT as _ };
    |                                                               ^^^^^^^^^^ not found in `libc`

error[E0425]: cannot find function `futimens` in crate `libc`
    --> C:\Users\n3xed\.rustup\toolchains\nightly-x86_64-pc-windows-msvc\lib\rustlib\src\rust\library\std\src\sys\unix\fs.rs:1114:36
     |
1114 |                 cvt(unsafe { libc::futimens(self.as_raw_fd(), times.0.as_ptr()) })?;
     |                                    ^^^^^^^^ help: a function with a similar name exists: `utimes`
     |
    ::: C:\Users\n3xed\.cargo\registry\src\github.com-1ecc6299db9ec823\libc-0.2.126\src\unix\mod.rs:1159:5
     |
1159 |     pub fn utimes(filename: *const ::c_char, times: *const ::timeval) -> ::c_int;
     |     ---------------------------------------------------------------------------- similarly named function `utimes` defined here
```

**Warnings**
```txt
warning: unused import: `crate::ffi::CStr`
 --> C:\Users\n3xed\.rustup\toolchains\nightly-x86_64-pc-windows-msvc\lib\rustlib\src\rust\library\std\src\sys\unix\mod.rs:3:5
  |
3 | use crate::ffi::CStr;
  |     ^^^^^^^^^^^^^^^^
  |
  = note: `#[warn(unused_imports)]` on by default

warning: unused import: `crate::ffi::CStr`
 --> C:\Users\n3xed\.rustup\toolchains\nightly-x86_64-pc-windows-msvc\lib\rustlib\src\rust\library\std\src\sys\unix\net.rs:2:5
  |
2 | use crate::ffi::CStr;
  |     ^^^^^^^^^^^^^^^^

warning: unused import: `crate::str`
 --> C:\Users\n3xed\.rustup\toolchains\nightly-x86_64-pc-windows-msvc\lib\rustlib\src\rust\library\std\src\sys\unix\net.rs:7:5
  |
7 | use crate::str;
  |     ^^^^^^^^^^

warning: unused import: `os`
 --> C:\Users\n3xed\.rustup\toolchains\nightly-x86_64-pc-windows-msvc\lib\rustlib\src\rust\library\std\src\sys\unix\thread.rs:7:18
  |
7 | use crate::sys::{os, stack_overflow};
  |                  ^^

warning: use of deprecated type alias `os::unix::raw::gid_t`: these type aliases are no longer supported by the standard library, the `libc` crate on crates.io should be used instead for the correct definitions
  --> C:\Users\n3xed\.rustup\toolchains\nightly-x86_64-pc-windows-msvc\lib\rustlib\src\rust\library\std\src\os\espidf\raw.rs:13:28
   |
13 | use crate::os::unix::raw::{gid_t, uid_t};
   |                            ^^^^^
   |
   = note: `#[warn(deprecated)]` on by default

warning: use of deprecated type alias `os::unix::raw::uid_t`: these type aliases are no longer supported by the standard library, the `libc` crate on crates.io should be used instead for the correct definitions
  --> C:\Users\n3xed\.rustup\toolchains\nightly-x86_64-pc-windows-msvc\lib\rustlib\src\rust\library\std\src\os\espidf\raw.rs:13:35
   |
13 | use crate::os::unix::raw::{gid_t, uid_t};
   |                                   ^^^^^

warning: use of deprecated type alias `os::espidf::raw::dev_t`: these type aliases are no longer supported by the standard library, the `libc` crate on crates.io should be used instead for the correct definitions
  --> C:\Users\n3xed\.rustup\toolchains\nightly-x86_64-pc-windows-msvc\lib\rustlib\src\rust\library\std\src\os\espidf\raw.rs:42:17
   |
42 |     pub st_dev: dev_t,
   |                 ^^^^^

warning: use of deprecated type alias `os::espidf::raw::ino_t`: these type aliases are no longer supported by the standard library, the `libc` crate on crates.io should be used instead for the correct definitions
  --> C:\Users\n3xed\.rustup\toolchains\nightly-x86_64-pc-windows-msvc\lib\rustlib\src\rust\library\std\src\os\espidf\raw.rs:44:17
   |
44 |     pub st_ino: ino_t,
   |                 ^^^^^

warning: use of deprecated type alias `os::espidf::raw::mode_t`: these type aliases are no longer supported by the standard library, the `libc` crate on crates.io should be used instead for the correct definitions
  --> C:\Users\n3xed\.rustup\toolchains\nightly-x86_64-pc-windows-msvc\lib\rustlib\src\rust\library\std\src\os\espidf\raw.rs:46:18
   |
46 |     pub st_mode: mode_t,
   |                  ^^^^^^

warning: use of deprecated type alias `os::espidf::raw::nlink_t`: these type aliases are no longer supported by the standard library, the `libc` crate on crates.io should be used instead for the correct definitions
  --> C:\Users\n3xed\.rustup\toolchains\nightly-x86_64-pc-windows-msvc\lib\rustlib\src\rust\library\std\src\os\espidf\raw.rs:48:19
   |
48 |     pub st_nlink: nlink_t,
   |                   ^^^^^^^

warning: use of deprecated type alias `os::unix::raw::uid_t`: these type aliases are no longer supported by the standard library, the `libc` crate on crates.io should be used instead for the correct definitions
  --> C:\Users\n3xed\.rustup\toolchains\nightly-x86_64-pc-windows-msvc\lib\rustlib\src\rust\library\std\src\os\espidf\raw.rs:50:17
   |
50 |     pub st_uid: uid_t,
   |                 ^^^^^

warning: use of deprecated type alias `os::unix::raw::gid_t`: these type aliases are no longer supported by the standard library, the `libc` crate on crates.io should be used instead for the correct definitions
  --> C:\Users\n3xed\.rustup\toolchains\nightly-x86_64-pc-windows-msvc\lib\rustlib\src\rust\library\std\src\os\espidf\raw.rs:52:17
   |
52 |     pub st_gid: gid_t,
   |                 ^^^^^

warning: use of deprecated type alias `os::espidf::raw::dev_t`: these type aliases are no longer supported by the standard library, the `libc` crate on crates.io should be used instead for the correct definitions
  --> C:\Users\n3xed\.rustup\toolchains\nightly-x86_64-pc-windows-msvc\lib\rustlib\src\rust\library\std\src\os\espidf\raw.rs:54:18
   |
54 |     pub st_rdev: dev_t,
   |                  ^^^^^

warning: use of deprecated type alias `os::espidf::raw::off_t`: these type aliases are no longer supported by the standard library, the `libc` crate on crates.io should be used instead for the correct definitions
  --> C:\Users\n3xed\.rustup\toolchains\nightly-x86_64-pc-windows-msvc\lib\rustlib\src\rust\library\std\src\os\espidf\raw.rs:56:18
   |
56 |     pub st_size: off_t,
   |                  ^^^^^

warning: use of deprecated type alias `os::espidf::raw::time_t`: these type aliases are no longer supported by the standard library, the `libc` crate on crates.io should be used instead for the correct definitions
  --> C:\Users\n3xed\.rustup\toolchains\nightly-x86_64-pc-windows-msvc\lib\rustlib\src\rust\library\std\src\os\espidf\raw.rs:58:19
   |
58 |     pub st_atime: time_t,
   |                   ^^^^^^

warning: use of deprecated type alias `os::espidf::raw::time_t`: these type aliases are no longer supported by the standard library, the `libc` crate on crates.io should be used instead for the correct definitions
  --> C:\Users\n3xed\.rustup\toolchains\nightly-x86_64-pc-windows-msvc\lib\rustlib\src\rust\library\std\src\os\espidf\raw.rs:60:19
   |
60 |     pub st_mtime: time_t,
   |                   ^^^^^^

warning: use of deprecated type alias `os::espidf::raw::time_t`: these type aliases are no longer supported by the standard library, the `libc` crate on crates.io should be used instead for the correct definitions
  --> C:\Users\n3xed\.rustup\toolchains\nightly-x86_64-pc-windows-msvc\lib\rustlib\src\rust\library\std\src\os\espidf\raw.rs:62:19
   |
62 |     pub st_ctime: time_t,
   |                   ^^^^^^

warning: use of deprecated type alias `os::espidf::raw::blksize_t`: these type aliases are no longer supported by the standard library, the `libc` crate on crates.io should be used instead for the correct definitions
  --> C:\Users\n3xed\.rustup\toolchains\nightly-x86_64-pc-windows-msvc\lib\rustlib\src\rust\library\std\src\os\espidf\raw.rs:64:21
   |
64 |     pub st_blksize: blksize_t,
   |                     ^^^^^^^^^

warning: use of deprecated type alias `os::espidf::raw::blkcnt_t`: these type aliases are no longer supported by the standard library, the `libc` crate on crates.io should be used instead for the correct definitions
  --> C:\Users\n3xed\.rustup\toolchains\nightly-x86_64-pc-windows-msvc\lib\rustlib\src\rust\library\std\src\os\espidf\raw.rs:66:20
   |
66 |     pub st_blocks: blkcnt_t,
   |                    ^^^^^^^^

```

</details>

ivmarkov, 2022-08-03:
I'm looking into it.

ivmarkov, 2022-08-03:
https://github.com/rust-lang/rust/pull/100119

MabezDev, 2022-08-05:
Thanks @ivmarkov!

closed 2022-08-05 by MabezDev