esp-rs/rust #135: Weird behavior with 1.63.0.2 on ESP32 with debug builds

bjoernQ, 2022-08-25:
The code in https://github.com/bjoernQ/miscompile prints 26.608885 when compiled in release mode on 1.63.0.2 but prints 0.015625 when compiled in debug mode.

Also, on 1.63.0.1 (and 1.63) it prints the correct  26.608885 in debug and release mode.

bjoernQ, 2022-08-26:
It is even reproducible with much less code:

```rust
...

    let foo = do_foo(3.14f64);
    let foo = foo as f32;
    println!("it is {}", foo);

    loop {}
}

#[inline(never)]
fn do_foo(param: f64) -> f64 {
    param * 2f64 / 3f64
}
```

This gives me 2.020576 (this seems to change randomly) on 1.63.0.2 debug but the correct 2.0933332 in release - always correct for 1.63

MabezDev, 2022-09-20:
This should now be fixed by the 1.64.0.0 release.

closed 2022-09-20 by MabezDev