esp-rs/rust #149: Weird behavior on variable assign inside a match case

FloppyDisck, 2022-11-12:
I tried this code:

```rust
pub enum TemperatureUnit {
    Celsius,
    Fahrenheit
}

let unit = TemperatureUnit::Fahrenheit;

    let (sub, mul) = match unit {
    TemperatureUnit::Celsius => (45f32, 175f32),
    TemperatureUnit::Fahrenheit => (49f32, 315f32),
};
    
println!("{} {}", sub, mul);
```

I expected to see this happen: Its supposed to print `49 315`

Instead, this happened: It prints `-197961600000000000000000000 -197961600000000000000000000`

Interesting enough, if you add `println!("");` inside the match cases, the output comes out fine...

### Meta

`rustc --version --verbose`:
```
rustc 1.62.1-nightly (e4f2cf605 2022-07-19)
binary: rustc
commit-hash: e4f2cf60561c2ad852af81cf1fddb7203ed058b1
commit-date: 2022-07-19
host: x86_64-unknown-linux-gnu
release: 1.62.1-nightly
LLVM version: 14.0.0

```

<details><summary>Backtrace</summary>
<p>

```
<backtrace>
```

</p>
</details>

usbalbin, 2022-11-13:
Could this be #136?

MabezDev, 2022-11-13:
> rustc 1.62

This issue was fixed back in 1.63, please upgrade your toolchain.

closed 2022-11-13 by MabezDev