esp-rs/rust #155: `f32::min` etc cause linker errors on bare-metal

bjoernQ, 2022-12-23:
Seems like that intrinsic generates code that wants to call `fminf` which is not present on bare-metal

It works fine with RISCV targets

MabezDev, 2023-02-22:
Do you remeber which target this was reproduced on?

Looks like for soft float targets we just need to enable it upstream, like it was done for riscv: https://github.com/rust-lang/compiler-builtins/pull/493. If this was reproduced on an ESP32 or ESP32S3 then we're missing the intrinsics in LLVM.

bjoernQ, 2023-02-22:
Ah I always wonder why people don't give hints on the used target and now I see I'm doing that, too.
But I remember it was on ESP32-S3 (at least some Xtensa target)

MabezDev, 2023-06-01:
With the 1.70 release, the compiler-builtins changes have been reflected! We can now close this.

To anyone running into duplicate symbol linker errors, you need to remove any of your own implementations in your crates/projects.

closed 2023-06-01 by MabezDev