esp-rs/rust #166: Build failure: setting llvm.experimental_targets is incompatible with download-ci-llvm

mbuesch, 2023-03-09:
With today's git default branch (4651851b1bb6f0d44543f0ec2c8d26e308aaadcf) I get the following build failure:

```
   ...
   Compiling fd-lock v3.0.8
   Compiling globset v0.4.8
   Compiling ignore v0.4.18
   Compiling toml v0.5.9
    Finished dev [unoptimized] target(s) in 1m 35s
thread 'main' panicked at 'setting llvm.experimental_targets is incompatible with download-ci-llvm.', config.rs:1207:17
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

My build commands are:

```
./configure --experimental-targets=Xtensa
./x.py build --stage 2
```

MabezDev, 2023-03-09:
You probably want to add `--release-channel=nightly` to your configure step, the default when building from source is `dev` which will try and download llvm artifacts.

Just incase you weren't aware, we have prebuilt toolchains for most operating systems available here: https://github.com/esp-rs/rust-build.

closed 2023-03-09 by MabezDev

mbuesch, 2023-03-10:
Thanks.
This fixed the build.