esp-rs/rust #190: FFT HW Acceleration

ramtej, 2023-08-24:
I am currently benchmarking some DSP routines on the ESP32 S3 platform in Rust. Several issues have already arisen, see https://github.com/esp-rs/rust/issues/180.

Upon reading the 'ESP32-S3 Technical Reference Manual', it became apparent that the S3 platform implements some SIMD as well as DSP operations in hardware, such as `EE.FFT.R2BF.S16` or `EE.CMUL.S16`. I would be willing to invest some time and and implement a hardware-accelerated FFT. I am more of a mathematician and do not know the ESP32 architecture well enough, so I need some support.

Would it therefore be possible for someone to guide me and show me where I need to start? I looked at the SHA HW acceleration and can understand most things, but not all.

Thanks, Jiri

ProfFan, 2024-07-20:
Bumping this issue as I was researching `memcpy` on ESP32-S3. It turns out that aligned memcpy with EE.VLD instructions can be 6 times faster than regular memcpy, quoting https://github.com/project-x51/esp32-s3-memorycopy:

```C
I (404) Memory Copy: Allocating 2 x 100kb in IRAM, alignment: 32 bytes
I (464) Memory Copy: 8-bit for loop copy IRAM->IRAM took 819922 CPU cycles = 28.59 MB/s
I (514) Memory Copy: 16-bit for loop copy IRAM->IRAM took 205776 CPU cycles = 113.90 MB/s
I (564) Memory Copy: 32-bit for loop copy IRAM->IRAM took 103383 CPU cycles = 226.71 MB/s
I (614) Memory Copy: 64-bit for loop copy IRAM->IRAM took 77682 CPU cycles = 301.71 MB/s
I (664) Memory Copy: memcpy IRAM->IRAM took 64323 CPU cycles = 364.37 MB/s
I (714) Memory Copy: async_memcpy IRAM->IRAM took 408520 CPU cycles = 57.37 MB/s
I (764) Memory Copy: PIE 128-bit (16 byte loop) IRAM->IRAM took 19498 CPU cycles = 1202.05 MB/s
I (814) Memory Copy: PIE 128-bit (32 byte loop) IRAM->IRAM took 13095 CPU cycles = 1789.81 MB/s
I (864) Memory Copy: DSP AES3 IRAM->IRAM took 15813 CPU cycles = 1482.17 MB/s
```

It would be great if we can have this in Rust (or is it already done?)

ProfFan, 2024-07-20:
> Sure, I didn't trust the objdump blindly, I made sure I tested it on one of my ESPS3S3 chips. I ran this memcpy test:
> 
> ```rust
> #[repr(align(16))]
> pub struct AlignedArray<const N: usize>([u8; N]);
> 
> #[inline(never)]
> pub unsafe fn aligned_memcpy_test<const N: usize>(dst: &mut AlignedArray<N>, src: &AlignedArray<N>) {
>     let src_addr = src.0.as_ptr();
>     let dst_addr = dst.0.as_mut_ptr();
>     assert!(src_addr.is_aligned_to(16));
>     assert!(dst_addr.is_aligned_to(16));
>     assert_eq!(N % 32, 0);
>     for _ in 0..N / 32 {
>         core::arch::asm!(
>             r#"
>                 ee.vld.128.ip    q0,  {src_addr},  16
>                 ee.vld.128.ip    q1,  {src_addr},  16
>                 ee.vst.128.ip    q0,  {dst_addr},  16
>                 ee.vst.128.ip    q1,  {dst_addr},  16
>             "#,
>             src_addr = in(reg) src_addr,
>             dst_addr = in(reg) dst_addr,
>         );
>     }
> }
> ```
> 
> src: [0, 1, 2, 3, 4, 5, 6, 7, ...., 0, 1, 2, 3, 4, 5,...] dst: [0, 1, 2, 3, ..., 14, 15, 0, 0, 0 ..., 0, 1, 2, 3, ... , 14, 15, ..., 0, 0, 0, ...] So the loop is jumping with an offset 320 instead of 32, just like the objdump.

Just tested this on current `esp-rs/rust` and it seems to work well.

Noxime, 2024-08-30:
Hi, I'm attempting to use the SIMD instructions with latest 1.80 release. Most instructions work as they're intended, but I've encountered a number of misassemblies especially on the arithmeric+load/store instructions. As an example:
```rs
asm!(
        "NOP",
        "NOP",
        "EE.VADDS.S8.LD.INCP q0, a15, q1, q2, q3",
        "NOP",
        "NOP",
)
```
Ends up as 
```obj
420876c5:	0020f0               	nop
420876c8:	0020f0               	nop
420876cb:	cf                      	.byte	0xcf
420876cc:	0299                	s32i.n	a9, a2, 0
420876ce:	f01c                	movi.n	a0, 31
420876d0:	f00020               	subx8	a0, a0, a2
420876d3:	f00020               	subx8	a0, a0, a2
```
(as disassembled by `xtensa-esp-elf-objdump`). As you can see, the instruction bytes are quite wrong, and executing does lead to IllegalInstruction exceptions.

So far I've observed this for instructions in the `*.LD/ST.INCP` group at least, but I would not be surprised if more were broken.