esp-rs/rust #197: LLVM ERROR: Error while trying to spill A4 from class AR: Cannot scavenge register without an emergency spill slot!

dimpolo, 2023-09-25:
Compiling a closed source project for the ESP32-S3 with `--release` leads to:
```text
LLVM ERROR: Error while trying to spill A4 from class AR: Cannot scavenge register without an emergency spill slot!
```

Is there anything I could do to help diagnose this?

### Meta

`rustc --version --verbose`:
```
rustc 1.72.1-nightly (edc37d22d 2023-09-15) (1.72.1.0)
binary: rustc
commit-hash: edc37d22db1cc1b112b91addeb1f79951c58e661
commit-date: 2023-09-15
host: x86_64-pc-windows-msvc
release: 1.72.1-nightly
LLVM version: 16.0.4
```

`Cargo.toml`
```toml
[profile.release]
codegen-units = 1
debug = true
```

`.cargo/config.toml:`
```toml
[build]
rustflags = [
    "-C", "link-arg=-nostartfiles",
    "-C", "link-arg=-Wl,-Tlinkall.x",
    "-C", "force-frame-pointers",
]
target = "xtensa-esp32s3-none-elf"

[unstable]
build-std = ["core"]
```

MabezDev, 2023-09-25:
> "-C", "force-frame-pointers",

This flag serves no purpose with Xtensa, there is an ISA-defined frame pointer. That said, it shouldn't error with it enabled. I think there was a bug in the template generator that included this. Does it still error without this flag?

Would you be able to try and produce a small test case where the error is produced?

dimpolo, 2023-09-25:
> "-C", "force-frame-pointers",

Removing this flag solved the error! Thanks for the quick response :)

closed 2023-09-25 by dimpolo

dimpolo, 2023-09-25:
@MabezDev 
Btw this is where I got the flag idea from. Should the text be updated?
https://github.com/esp-rs/esp-backtrace#esp-backtrace---backtrace-for-esp32-bare-metal

MabezDev, 2023-09-25:
Ooh, yeah that should say only for the RISCV targets. Would you mind filing an issue/PR?