esp-rs/rust #203: With Esp projects only - Rust Analyzer is not working properly

yanshay, 2023-11-03:
I'm using Neovim (based on LazyVim) for Rust development and with standard projects everything work properly including Rust Analyzer.

When I work with Esp-Idf embedded projects Rust Analyzer is only partially working:
1. Syntax errors are shown
2. Completion work as well
3. But some errors don't show, for example, in the below code it doesn't show that tt() function is missing. However, it is colored red (signaling there is an error), and if I try to use code action to fix the issue (that I know exists) it knows to suggest adding the function.
```
fn main() {
    esp_idf_svc::sys::link_patches();

    esp_idf_svc::log::EspLogger::initialize_default();

    log::info!("Hello, world!");
    tt();
}
```
If I run cargo check from command line I get the proper error message.

Any ideas what could cause this and/or how can I troubleshoot it?

MabezDev, 2023-11-03:
Can you obtain the error from rust analyzer?

yanshay, 2023-11-03:
> Can you obtain the error from rust analyzer?

What do you mean by that exactly? In Nvim, which use Rust Analyzer, the error doesn't appear as diagnostics or in any other place where those errors usually appear.
If there is some manual way to use rust analyzer then I'm not aware of that, is there?

MabezDev, 2023-11-03:
I don't use neovim (for Rust at least) but there must be a way to obtain the error from rust-analzyer within neovim. In vscode there is a separate log window.

yanshay, 2023-11-03:
> I don't use neovim (for Rust at least) but there must be a way to obtain the error from rust-analzyer within neovim. In vscode there is a separate log window.

Yes, when rust analyzer reports errors, there are several ways I see it (as popup window, or another window or interleaved with the code itself).
Syntax errors are shown in all those ways, the error I point to above (as well as others) doen't appear in any of these places. They are only shown by the code turning red. When I fix the error (because I see there's an error from running cargo check in command line) the color turns to the right syntax highlighting color.
That's exactly the issue I have - some of the errors don't show up. And that's only in Esp projects. In other project types those errors do appear.

yanshay, 2023-11-03:
I pinpointed the cause on a clean project from esp-idf template to `pio = ["esp-idf-svc/pio"]` in Cargo.toml.
If it's there then the compile error doesn't show (so faulty), if it isn't there then at least the behavior is as expected.
I'm new to Esp programming, I believe pio stands for PlatformIO. 
Does that make sense that it doesn't work well with it? When is this pio required in the first place?

```
[package]
name = "esp-prob"
version = "0.1.0"
edition = "2021"
resolver = "2"
rust-version = "1.71"

[profile.release]
opt-level = "s"

[profile.dev]
debug = true    # Symbols are nice and they don't increase the size on Flash
opt-level = "z"

[features]
default = ["std", "embassy", "esp-idf-svc/native"]

# -> below line, pio, cause the issues
# pio = ["esp-idf-svc/pio"]
std = ["alloc", "esp-idf-svc/binstart", "esp-idf-svc/std"]
alloc = ["esp-idf-svc/alloc"]
nightly = ["esp-idf-svc/nightly"]
experimental = ["esp-idf-svc/experimental"]
embassy = ["esp-idf-svc/embassy-sync", "esp-idf-svc/critical-section", "esp-idf-svc/embassy-time-driver"]

[dependencies]
# log = { version = "0.4", default-features = false }
esp-idf-svc = { version = "0.47.1", default-features = false }

[build-dependencies]
embuild = "0.31.3"
```

ivmarkov, 2024-01-21:
The reason why the `pio` line was a problem is because - under certain setups - Rust Analyzer does `cargo check --all-features`.

Well, `--all features` was a problem on `esp-idf-sys` as it did have features which are mutually incompatible, hence enabling **all** features actually did fail the build (try enabling both feature `pio` and `native` on the latest-released `esp-idf-sys` from crates.io).
Ditto for `esp-idf-hal` - which - up until recently was having a similar conflict between features `riscv-ulp-hal` and `esp-idf-sys` (and did not work with `--no-default-features` either, unless you explicitly enabled the `esp-idf-sys` feature).

All of the above is now addressed in the GIT version of the above crates (to be released soon) in that both `--all-features` and `--no-default-features` work fine.

As a temporary workaround, either do what you do, or disable `--all-features` in the `cargo check` line Rust Analyzer uses.

@MabezDev As per the fixes mentioned above ^^^ we should close this.

closed 2024-10-23 by MabezDev