esp-rs/rust #227: Rustc fails to run on Mac OS 11.7.7 (20G1345)

isaac-asdf, 2024-06-27:
I followed the install instructions from the Rust on ESP book, using `espup` and I created a new repo from the esp-template. When I try to `cargo check` the process errors out. It seems like an OS version issue. Is it possible to install an older version of the Rust fork through `espup`? I had been able to run this same process ~6months ago, but after updating with `espup update` it no longer works.

```
% cargo check
error: process didn't exit successfully: `/Users/me/.rustup/toolchains/esp/bin/rustc -vV` (signal: 6, SIGABRT: process abort signal)
--- stderr
dyld: Symbol not found: __ZNKSt3__115basic_stringbufIcNS_11char_traitsIcEENS_9allocatorIcEEE3strEv
  Referenced from: /Users/me/.rustup/toolchains/esp/bin/../lib/librustc_driver-c61a62fc4508abaa.dylib (which was built for Mac OS X 12.0)
  Expected in: /usr/lib/libc++.1.dylib
```

MabezDev, 2024-10-28:
Could you try the latest toolchain?

What version of macos are you running?

isaac-asdf, 2024-10-30:
It seems this is still an issue. I ran `espup --update` and then `cargo check` and got the same error. I'm on macos 11.7.10

MabezDev, 2024-10-30:
macos 11 is quite old and no longer maintained according to https://en.wikipedia.org/wiki/MacOS. I suggest updating your system as we'll likely be updating our builder soon too.

closed 2024-10-30 by MabezDev