esp-rs/rust #236: Strange compiling error for xTensa

trinhanhtuan, 2024-09-24:
Hi,
I meet a strange issue while compiling for ESP32 even with the bare minimal hello-world auto-generated by `cargo generate`, here is the errors:
```
   Compiling spin v0.9.8
error: rustc interrupted by SIGSEGV, printing backtrace

/root/.rustup/toolchains/esp/lib/librustc_driver-11f442b45f11e453.so(+0x386e6c6)[0x7fb683dda6c6]
/lib/x86_64-linux-gnu/libc.so.6(+0x45320)[0x7fb680240320]
/home/PrvProjects/ESP32/helloworld-esp32/target/release/deps/libdarling_macro-54b0a8e73562b65b.so(+0x30ca55)[0x7fb67cc1ca55]
/home/PrvProjects/ESP32/helloworld-esp32/target/release/deps/libdarling_macro-54b0a8e73562b65b.so(+0x2353c7)[0x7fb67cb453c7]
/home/PrvProjects/ESP32/helloworld-esp32/target/release/deps/libdarling_macro-54b0a8e73562b65b.so(+0x1f0acd)[0x7fb67cb00acd]
/home/PrvProjects/ESP32/helloworld-esp32/target/release/deps/libdarling_macro-54b0a8e73562b65b.so(+0x1f0a26)[0x7fb67cb00a26]
/home/PrvProjects/ESP32/helloworld-esp32/target/release/deps/libdarling_macro-54b0a8e73562b65b.so(+0x1f1c4c)[0x7fb67cb01c4c]
/home/PrvProjects/ESP32/helloworld-esp32/target/release/deps/libdarling_macro-54b0a8e73562b65b.so(+0x285e0e)[0x7fb67cb95e0e]
/home/PrvProjects/ESP32/helloworld-esp32/target/release/deps/libdarling_macro-54b0a8e73562b65b.so(+0x244450)[0x7fb67cb54450]
/home/PrvProjects/ESP32/helloworld-esp32/target/release/deps/libdarling_macro-54b0a8e73562b65b.so(+0x28ac98)[0x7fb67cb9ac98]
/home/PrvProjects/ESP32/helloworld-esp32/target/release/deps/libdarling_macro-54b0a8e73562b65b.so(+0x243fb7)[0x7fb67cb53fb7]
/home/PrvProjects/ESP32/helloworld-esp32/target/release/deps/libdarling_macro-54b0a8e73562b65b.so(+0x2526be)[0x7fb67cb626be]
/home/PrvProjects/ESP32/helloworld-esp32/target/release/deps/libdarling_macro-54b0a8e73562b65b.so(+0x245ae5)[0x7fb67cb55ae5]
/home/PrvProjects/ESP32/helloworld-esp32/target/release/deps/libdarling_macro-54b0a8e73562b65b.so(+0x25268a)[0x7fb67cb6268a]
/home/PrvProjects/ESP32/helloworld-esp32/target/release/deps/libdarling_macro-54b0a8e73562b65b.so(+0x289dcb)[0x7fb67cb99dcb]
/home/PrvProjects/ESP32/helloworld-esp32/target/release/deps/libdarling_macro-54b0a8e73562b65b.so(+0x1e29f2)[0x7fb67caf29f2]
/home/PrvProjects/ESP32/helloworld-esp32/target/release/deps/libdarling_macro-54b0a8e73562b65b.so(+0x287677)[0x7fb67cb97677]
/home/PrvProjects/ESP32/helloworld-esp32/target/release/deps/libdarling_macro-54b0a8e73562b65b.so(+0x1e4377)[0x7fb67caf4377]
/home/PrvProjects/ESP32/helloworld-esp32/target/release/deps/libdarling_macro-54b0a8e73562b65b.so(+0x1e485f)[0x7fb67caf485f]
/home/PrvProjects/ESP32/helloworld-esp32/target/release/deps/libdarling_macro-54b0a8e73562b65b.so(+0x286867)[0x7fb67cb96867]
/home/PrvProjects/ESP32/helloworld-esp32/target/release/deps/libdarling_macro-54b0a8e73562b65b.so(+0x1e50bc)[0x7fb67caf50bc]
/home/PrvProjects/ESP32/helloworld-esp32/target/release/deps/libdarling_macro-54b0a8e73562b65b.so(+0x1e5b2b)[0x7fb67caf5b2b]
/home/PrvProjects/ESP32/helloworld-esp32/target/release/deps/libdarling_macro-54b0a8e73562b65b.so(+0x28d444)[0x7fb67cb9d444]
/home/PrvProjects/ESP32/helloworld-esp32/target/release/deps/libdarling_macro-54b0a8e73562b65b.so(+0x24f0ac)[0x7fb67cb5f0ac]
/home/PrvProjects/ESP32/helloworld-esp32/target/release/deps/libdarling_macro-54b0a8e73562b65b.so(+0x288937)[0x7fb67cb98937]
/home/PrvProjects/ESP32/helloworld-esp32/target/release/deps/libdarling_macro-54b0a8e73562b65b.so(+0x247fd1)[0x7fb67cb57fd1]
/home/PrvProjects/ESP32/helloworld-esp32/target/release/deps/libdarling_macro-54b0a8e73562b65b.so(+0x247816)[0x7fb67cb57816]
/home/PrvProjects/ESP32/helloworld-esp32/target/release/deps/libdarling_macro-54b0a8e73562b65b.so(+0x288637)[0x7fb67cb98637]
/home/PrvProjects/ESP32/helloworld-esp32/target/release/deps/libdarling_macro-54b0a8e73562b65b.so(+0x23de4c)[0x7fb67cb4de4c]
/home/PrvProjects/ESP32/helloworld-esp32/target/release/deps/libdarling_macro-54b0a8e73562b65b.so(+0x298581)[0x7fb67cba8581]
/home/PrvProjects/ESP32/helloworld-esp32/target/release/deps/libdarling_macro-54b0a8e73562b65b.so(+0x28a70c)[0x7fb67cb9a70c]
/home/PrvProjects/ESP32/helloworld-esp32/target/release/deps/libdarling_macro-54b0a8e73562b65b.so(+0x23d54a)[0x7fb67cb4d54a]
/home/PrvProjects/ESP32/helloworld-esp32/target/release/deps/libdarling_macro-54b0a8e73562b65b.so(+0x2881e7)[0x7fb67cb981e7]
/home/PrvProjects/ESP32/helloworld-esp32/target/release/deps/libdarling_macro-54b0a8e73562b65b.so(+0x2700de)[0x7fb67cb800de]
/home/PrvProjects/ESP32/helloworld-esp32/target/release/deps/libdarling_macro-54b0a8e73562b65b.so(+0x20fdae)[0x7fb67cb1fdae]
/home/PrvProjects/ESP32/helloworld-esp32/target/release/deps/libdarling_macro-54b0a8e73562b65b.so(+0xe0b9b)[0x7fb67c9f0b9b]
/home/PrvProjects/ESP32/helloworld-esp32/target/release/deps/libdarling_macro-54b0a8e73562b65b.so(+0xe074d)[0x7fb67c9f074d]
/home/PrvProjects/ESP32/helloworld-esp32/target/release/deps/libdarling_macro-54b0a8e73562b65b.so(+0xe0647)[0x7fb67c9f0647]
/home/PrvProjects/ESP32/helloworld-esp32/target/release/deps/libdarling_macro-54b0a8e73562b65b.so(+0xe05e5)[0x7fb67c9f05e5]
/home/PrvProjects/ESP32/helloworld-esp32/target/release/deps/libdarling_macro-54b0a8e73562b65b.so(+0xe1ccd)[0x7fb67c9f1ccd]
/home/PrvProjects/ESP32/helloworld-esp32/target/release/deps/libdarling_macro-54b0a8e73562b65b.so(+0xe0ad6)[0x7fb67c9f0ad6]
/home/PrvProjects/ESP32/helloworld-esp32/target/release/deps/libdarling_macro-54b0a8e73562b65b.so(+0xddcea)[0x7fb67c9edcea]
/home/PrvProjects/ESP32/helloworld-esp32/target/release/deps/libdarling_macro-54b0a8e73562b65b.so(+0xddaa2)[0x7fb67c9edaa2]
/home/PrvProjects/ESP32/helloworld-esp32/target/release/deps/libdarling_macro-54b0a8e73562b65b.so(+0xdf04a)[0x7fb67c9ef04a]
/home/PrvProjects/ESP32/helloworld-esp32/target/release/deps/libdarling_macro-54b0a8e73562b65b.so(+0xdcdd5)[0x7fb67c9ecdd5]
/home/PrvProjects/ESP32/helloworld-esp32/target/release/deps/libdarling_macro-54b0a8e73562b65b.so(+0xdf7f0)[0x7fb67c9ef7f0]
/home/PrvProjects/ESP32/helloworld-esp32/target/release/deps/libdarling_macro-54b0a8e73562b65b.so(+0xe01f4)[0x7fb67c9f01f4]
/home/PrvProjects/ESP32/helloworld-esp32/target/release/deps/libdarling_macro-54b0a8e73562b65b.so(+0xdddcb)[0x7fb67c9eddcb]
/home/PrvProjects/ESP32/helloworld-esp32/target/release/deps/libdarling_macro-54b0a8e73562b65b.so(+0xda14c)[0x7fb67c9ea14c]
/home/PrvProjects/ESP32/helloworld-esp32/target/release/deps/libdarling_macro-54b0a8e73562b65b.so(+0xddc71)[0x7fb67c9edc71]
/home/PrvProjects/ESP32/helloworld-esp32/target/release/deps/libdarling_macro-54b0a8e73562b65b.so(+0xde3e2)[0x7fb67c9ee3e2]
/root/.rustup/toolchains/esp/lib/librustc_driver-11f442b45f11e453.so(+0x9a8586d)[0x7fb689ff186d]
/root/.rustup/toolchains/esp/lib/librustc_driver-11f442b45f11e453.so(+0x9a845c3)[0x7fb689ff05c3]
/root/.rustup/toolchains/esp/lib/librustc_driver-11f442b45f11e453.so(+0x9ad6b49)[0x7fb68a042b49]
/root/.rustup/toolchains/esp/lib/librustc_driver-11f442b45f11e453.so(_RNvXs1_NtCs8FqiUvy9Deh_12rustc_expand10proc_macroNtB5_15DeriveProcMacroNtNtB7_4base17MultiItemModifier6expand+0x11c)[0x7fb68a06e85c]
/root/.rustup/toolchains/esp/lib/librustc_driver-11f442b45f11e453.so(_RNvMs1_NtCs8FqiUvy9Deh_12rustc_expand6expandNtB5_13MacroExpander21fully_expand_fragment+0x128a)[0x7fb689fbbbda]
/root/.rustup/toolchains/esp/lib/librustc_driver-11f442b45f11e453.so(_RNvMs1_NtCs8FqiUvy9Deh_12rustc_expand6expandNtB5_13MacroExpander12expand_crate+0x413)[0x7fb689fba6c3]
/root/.rustup/toolchains/esp/lib/librustc_driver-11f442b45f11e453.so(+0x3bc5f4d)[0x7fb684131f4d]
/root/.rustup/toolchains/esp/lib/librustc_driver-11f442b45f11e453.so(_RNvNtCshkBckzDuHTW_15rustc_interface6passes25resolver_for_lowering_raw+0x874)[0x7fb6840d0c84]
/root/.rustup/toolchains/esp/lib/librustc_driver-11f442b45f11e453.so(+0x8994325)[0x7fb688f00325]
/root/.rustup/toolchains/esp/lib/librustc_driver-11f442b45f11e453.so(+0x8d4963c)[0x7fb6892b563c]
/root/.rustup/toolchains/esp/lib/librustc_driver-11f442b45f11e453.so(+0x89e818a)[0x7fb688f5418a]
/root/.rustup/toolchains/esp/lib/librustc_driver-11f442b45f11e453.so(+0x8cdc5b0)[0x7fb6892485b0]
/root/.rustup/toolchains/esp/lib/librustc_driver-11f442b45f11e453.so(_RNvMsE_NtNtCs9OHcTqa5xde_12rustc_middle2ty7contextNtB5_6TyCtxt21resolver_for_lowering+0x3c)[0x7fb68a358fec]
[... 24 lines cut ...]
rustup 1.27.1 (54dd3d00f 2024-04-24)
info: This is the version for the rustup toolchain manager, not the rustc compiler.
info: The currently active `rustc` version is `rustc 1.81.0 (eeb90cda1 2024-09-04)`
# cargo --version
cargo 1.81.0 (2dbb1af80 2024-08-20)
# rustc --version
rustc 1.81.0 (eeb90cda1 2024-09-04)
# espup --version
espup 0.12.2
# cat /etc/os-release
PRETTY_NAME="Ubuntu 24.04.1 LTS"
NAME="Ubuntu"
VERSION_ID="24.04"
VERSION="24.04.1 LTS (Noble Numbat)"
VERSION_CODENAME=noble
ID=ubuntu
ID_LIKE=debian
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
UBUNTU_CODENAME=noble
LOGO=ubuntu-logo
#
```

MabezDev, 2024-09-24:
What's the output of `rustc +esp --version --verbose`?

trinhanhtuan, 2024-09-25:
Hi,
I've just compiled an auto-generated hello-world for ESP32-C6, ESP32-S3, ESP32-S2 successfully. Apparently the error is only happened with ESP32 xTensa LX6.

trinhanhtuan, 2024-10-04:
> What's the output of `rustc +esp --version --verbose`?

Not quite sure if this question comes to right place but I still try. Do you know when this bug will be fixed?

Thank you,
Tuan

MabezDev, 2024-10-28:
Could you try with the latest 1.82 release? We just updated the LLVM version which might fix this.

trinhanhtuan, 2024-10-29:
> Could you try with the latest 1.82 release? We just updated the LLVM version which might fix this.

Yah, I did it few days ago, it works fine. Sorry not to notify this thread.

MabezDev, 2024-10-29:
Glad it's working now!

closed 2024-10-29 by MabezDev