esp-rs/rust #247: Support for position independent code (PIC)

benhansen-io, 2024-12-05:
I would like to use [espressif/elf_loader](https://components.espressif.com/components/espressif/elf_loader) to load an elf file with Rust code on the esp32s3 (xtensa) however elf loader currently requires position independent code. When I try to compile Rust code with the esp-rs compiler using the rust flag: -C relocation-model=pic

I get the following error message:

```
LLVM ERROR: PIC relocations is not supported
error: could not compile `core` (lib)
```

MabezDev, 2024-12-05:
PIC is dependent solely on LLVM support, could you open the issue under https://github.com/espressif/llvm-project instead?

benhansen-io, 2024-12-05:
Done. Issue is https://github.com/espressif/llvm-project/issues/106

closed 2024-12-05 by benhansen-io