esp-rs/rust #249: Signedness mismatch between core::ffi::c_char and libc::c_char

jothan, 2024-12-21:
I tried this code:

```rust
unsafe { libc::printf(c"%s".as_ptr(), line.as_ptr()) };
```

I expected to be able to compile the above code, but the compiler had a different opinion:

```rust
error[E0308]: mismatched types
  --> src/logging.rs:90:27
   |
90 |     unsafe { libc::printf(c"%s".as_ptr(), line.as_ptr()) };
   |              ------------ ^^^^^^^^^^^^^^ expected `*const u8`, found `*const i8`
   |              |
   |              arguments to this function are incorrect
   |
   = note: expected raw pointer `*const u8`
              found raw pointer `*const i8`
note: function defined here
  --> index.crates.io-6f17d22bba15001f/libc-0.2.169/src/unix/mod.rs:694:12
```

It would seem that the latest libc 0.2.169 flipped its signedness of c_char from signed to unsigned for ESP-IDF:
https://github.com/rust-lang/libc/commit/c66faeba57869080a3fded445acdb5a42116f628

rustc should probably match this signedness change on ESP-IDF and update the definition somewhere around here: https://doc.rust-lang.org/beta/src/core/ffi/mod.rs.html#52

Doing a bit of searching around, I am about 80% sure that char is unsigned on the C side on ESP-IDF.

### Meta
`rustc --version --verbose`:
```
rustc 1.83.0-nightly (103cde010 2024-11-26) (1.83.0.0)
binary: rustc
commit-hash: 103cde01096863220765fd2e54519d011a5740ac
commit-date: 2024-11-26
host: x86_64-unknown-linux-gnu
release: 1.83.0-nightly
LLVM version: 18.1.2
```

MabezDev, 2025-01-24:
Sorry, this got lost in my notifications over the holidays. We fixed this in https://github.com/rust-lang/libc/pull/4195 and I believe libc has had a release since then, so this should be fixed. Please re open if that's not the case.

closed 2025-01-24 by MabezDev

jothan, 2025-01-24:
> Sorry, this got lost in my notifications over the holidays. We fixed this in [rust-lang/libc#4195](https://github.com/rust-lang/libc/pull/4195) and I believe libc has had a release since then, so this should be fixed. Please re open if that's not the case.

Ok, looks like the rustc side will match once 1.85 is out. I'm going to test it then.

rustc PR: https://github.com/rust-lang/rust/pull/132975/files

jothan, 2025-01-24:
Big thanks 🙏🏼👍🏼❤️