esp-rs/rust #253: error: Undefined temporary symbol .LBB11_19 when compiling rustls

nbc12, 2025-02-07:
I'm trying to get an FTPS client working on an esp32. I have to use TLS/SSL for my application.
I'm compiling it for xtensa for the regular esp32-wroom-32. I'm using Ubuntu on WSL on x86-64, and I just installed the esp-rs/rust compiler yesterday.
I'm using the suppaftp crate because it supports rustls so it is a great fit for the esp32.

If you just try and build the suppaftp crate, it gives a few errors. This is because the ring crate says:
`error[E0277]: the trait bound `SystemRandom: ring::rand::SecureRandom` is not satisfied`
You can fix this with a feature flag in the ring crate: `less-safe-getrandom-espidf`.
I cloned the suppaftp and rustls crates locally and enabled the feature in rustls's `Cargo.toml`.
Note that for some reason I also had to clone ring locally, I'm not entirely sure why.
This fixes all the compile errors.

Then, running `cargo build` again, rustc gives an error:
```
   Compiling rustls v0.23.22 (/home/noah/esp-rs-repro/deps/rustls/rustls)
   Compiling esp-rs-repro v0.1.0 (/home/noah/esp-rs-repro)
   Compiling suppaftp v6.0.7 (/home/noah/esp-rs-repro/deps/suppaftp/suppaftp)
error: Undefined temporary symbol .LBB11_19

error: could not compile `rustls` (lib) due to 1 previous error
```

This was run with opt-level = "z".

The build supposedly suceeds on opt level 0, but every other level it errors out.
I haven't verified if the program actually works as intended on opt level 0 as I do not have access to the network with the FTP server at the moment, but it doesn't give any errors.

For opt level 2 and 3, it gives two:
```
error: Undefined temporary symbol .LBB6_29
error: Undefined temporary symbol .LBB6_1
```

This bug occurs when compiling an empty main function.

suppaftp imports rustls like this: `rustls-crate = { package = "rustls", version = "^0.23", default-features = false, features = ["logging", "ring", "std", "tls12"], optional = true }`
However if the bin crate imports rustls using the same line, the problem is not present, so it's something about the combination of rustls and suppaftp that causes the bug to occur.

# Minimum reproduction
Admittedly I'm out of my depth here, so I don't know how to minimize this more.

Reproduction:
I have uploaded my crate to github here, though I don't know enough about git to make my changes to the dependencies persist in my commit.
https://github.com/nbc12/esp-rs-repro

Because of this, I think you're better off reproducing it by doing these steps:

1. Create a new esp-idf template using `cargo generate esp-rs/esp-idf-template cargo`. (The esp-rs rust book talks about this in detail if you need.) Select the esp32 target. Do not configure advanced template options.
2. In the root of the new bin crate:
```
mkdir deps
git clone https://github.com/rustls/rustls.git deps/rustls
git clone https://github.com/briansmith/ring.git deps/ring
git clone https://github.com/veeso/suppaftp.git deps/suppaftp
```
3. in `deps/rustls/Cargo.toml`, replace ring with this line: `ring = { path = "../ring", features = ["less-safe-getrandom-espidf"]}`
4. in `deps/suppaftp/suppaftp/Cargo.toml`, replace rustls-crate with this line: `rustls-crate = { package = "rustls", path="../../rustls/rustls", default-features = false, features = ["logging", "ring", "std", "tls12"], optional = true }`
5. in `./Cargo.toml`, add `suppaftp = { path = "./deps/suppaftp/suppaftp", features = ["rustls"] }`
6. `cargo build`

Output:
```
...
   Compiling futures-lite v2.6.0
   Compiling aho-corasick v1.1.3
   Compiling chrono v0.4.39
warning: use of deprecated macro `prefixed_export`: `#[export_name]` creates problems and we will stop doing it.
   --> deps/ring/src/arithmetic/montgomery.rs:191:1
    |
191 | prefixed_export! {
    | ^^^^^^^^^^^^^^^
    |
    = note: `#[warn(deprecated)]` on by default

   Compiling regex-automata v0.4.9
   Compiling rustls-webpki v0.102.8
warning: `ring` (lib) generated 1 warning
   Compiling regex v1.11.1
   Compiling lazy-regex v3.4.1
   Compiling suppaftp v6.0.7 (/home/noah/asdf/deps/suppaftp/suppaftp)
error: Undefined temporary symbol .LBB11_19

error: could not compile `rustls` (lib) due to 1 previous error
```

esp-idf-sys takes forever to do a clean build, just wait for it.

Setting RUSTC_BACKTRACE=1 does not change the output.

Output of `rustc --version --verbose`:
```
rustc 1.84.0-nightly (b0bd11bcc 2025-01-10) (1.84.0.0)
binary: rustc
commit-hash: b0bd11bcc4e5baf48a4fa8660e360e4a8248966b
commit-date: 2025-01-10
host: x86_64-unknown-linux-gnu
release: 1.84.0-nightly
LLVM version: 18.1.2
```

I'd love to help fix this if I can, I just have no idea where to start.

MabezDev, 2025-02-07:
Can you try compiling for a RISCV  esp std target too?

nbc12, 2025-02-07:
I followed the repro steps in my above post, except used the esp32-c3 template instead of the esp32 template.

```
 cargo +nightly build
   Compiling ring v0.17.8
   Compiling ring v0.17.8 (/home/noah/riscv-repro/deps/ring)
warning: ring@0.17.8: Compiler family detection failed due to error: ToolNotFound: Failed to find tool. Is `riscv32-esp-elf-gcc` installed?
warning: ring@0.17.8: Compiler family detection failed due to error: ToolNotFound: Failed to find tool. Is `riscv32-esp-elf-gcc` installed?
error: failed to run custom build command for `ring v0.17.8`

Caused by:
  process didn't exit successfully: `/home/noah/riscv-repro/target/debug/build/ring-d6c44fdf94a49c71/build-script-build` (exit status: 1)
  --- stdout
  cargo:rerun-if-env-changed=RING_PREGENERATE_ASM
  cargo:rustc-env=RING_CORE_PREFIX=ring_core_0_17_8_
  OPT_LEVEL = Some(z)
  OUT_DIR = Some(/home/noah/riscv-repro/target/riscv32imc-esp-espidf/debug/build/ring-18d1477061dda374/out)
  TARGET = Some(riscv32imc-esp-espidf)
  HOST = Some(x86_64-unknown-linux-gnu)
  cargo:rerun-if-env-changed=CC_riscv32imc-esp-espidf
  CC_riscv32imc-esp-espidf = None
  cargo:rerun-if-env-changed=CC_riscv32imc_esp_espidf
  CC_riscv32imc_esp_espidf = None
  cargo:rerun-if-env-changed=TARGET_CC
  TARGET_CC = None
  cargo:rerun-if-env-changed=CC
  CC = None
  cargo:rerun-if-env-changed=CROSS_COMPILE
  CROSS_COMPILE = None
  RUSTC_LINKER = Some(ldproxy)
  cargo:rerun-if-env-changed=CC_ENABLE_DEBUG_OUTPUT
  cargo:warning=Compiler family detection failed due to error: ToolNotFound: Failed to find tool. Is `riscv32-esp-elf-gcc` installed?
  RUSTC_WRAPPER = None
  cargo:rerun-if-env-changed=CRATE_CC_NO_DEFAULTS
  CRATE_CC_NO_DEFAULTS = None
  DEBUG = Some(true)
  CARGO_CFG_TARGET_FEATURE = Some(c,m)
  cargo:rerun-if-env-changed=CFLAGS_riscv32imc-esp-espidf
  CFLAGS_riscv32imc-esp-espidf = None
  cargo:rerun-if-env-changed=CFLAGS_riscv32imc_esp_espidf
  CFLAGS_riscv32imc_esp_espidf = None
  cargo:rerun-if-env-changed=TARGET_CFLAGS
  TARGET_CFLAGS = None
  cargo:rerun-if-env-changed=CFLAGS
  CFLAGS = None
  CARGO_ENCODED_RUSTFLAGS = Some(--cfgespidf_time64)
  cargo:warning=Compiler family detection failed due to error: ToolNotFound: Failed to find tool. Is `riscv32-esp-elf-gcc` installed?

  --- stderr

  error occurred in cc-rs: Failed to find tool. Is `riscv32-esp-elf-gcc` installed?

warning: build failed, waiting for other jobs to finish...
warning: ring@0.17.8: Compiler family detection failed due to error: ToolNotFound: Failed to find tool. Is `riscv32-esp-elf-gcc` installed?
warning: ring@0.17.8: Compiler family detection failed due to error: ToolNotFound: Failed to find tool. Is `riscv32-esp-elf-gcc` installed?
error: failed to run custom build command for `ring v0.17.8 (/home/noah/riscv-repro/deps/ring)`

Caused by:
  process didn't exit successfully: `/home/noah/riscv-repro/target/debug/build/ring-4a8046c6c11c2518/build-script-build` (exit status: 1)
  --- stdout
  cargo:rerun-if-env-changed=CARGO_MANIFEST_DIR
  cargo:rerun-if-env-changed=CARGO_PKG_NAME
  cargo:rerun-if-env-changed=CARGO_PKG_VERSION_MAJOR
  cargo:rerun-if-env-changed=CARGO_PKG_VERSION_MINOR
  cargo:rerun-if-env-changed=CARGO_PKG_VERSION_PATCH
  cargo:rerun-if-env-changed=CARGO_PKG_VERSION_PRE
  cargo:rerun-if-env-changed=CARGO_MANIFEST_LINKS
  cargo:rerun-if-env-changed=RING_PREGENERATE_ASM
  cargo:rerun-if-env-changed=OUT_DIR
  cargo:rerun-if-env-changed=CARGO_CFG_TARGET_ARCH
  cargo:rerun-if-env-changed=CARGO_CFG_TARGET_OS
  cargo:rerun-if-env-changed=CARGO_CFG_TARGET_ENV
  cargo:rerun-if-env-changed=CARGO_CFG_TARGET_ENDIAN
  cargo:rerun-if-env-changed=DEBUG
  cargo:rerun-if-env-changed=PERL_EXECUTABLE
  OPT_LEVEL = Some(z)
  OUT_DIR = Some(/home/noah/riscv-repro/target/riscv32imc-esp-espidf/debug/build/ring-de2aa6abdc106f42/out)
  TARGET = Some(riscv32imc-esp-espidf)
  HOST = Some(x86_64-unknown-linux-gnu)
  cargo:rerun-if-env-changed=CC_riscv32imc-esp-espidf
  CC_riscv32imc-esp-espidf = None
  cargo:rerun-if-env-changed=CC_riscv32imc_esp_espidf
  CC_riscv32imc_esp_espidf = None
  cargo:rerun-if-env-changed=TARGET_CC
  TARGET_CC = None
  cargo:rerun-if-env-changed=CC
  CC = None
  cargo:rerun-if-env-changed=CROSS_COMPILE
  CROSS_COMPILE = None
  RUSTC_LINKER = Some(ldproxy)
  cargo:rerun-if-env-changed=CC_ENABLE_DEBUG_OUTPUT
  cargo:warning=Compiler family detection failed due to error: ToolNotFound: Failed to find tool. Is `riscv32-esp-elf-gcc` installed?
  RUSTC_WRAPPER = None
  cargo:rerun-if-env-changed=CRATE_CC_NO_DEFAULTS
  CRATE_CC_NO_DEFAULTS = None
  DEBUG = Some(true)
  CARGO_CFG_TARGET_FEATURE = Some(c,m)
  cargo:rerun-if-env-changed=CFLAGS_riscv32imc-esp-espidf
  CFLAGS_riscv32imc-esp-espidf = None
  cargo:rerun-if-env-changed=CFLAGS_riscv32imc_esp_espidf
  CFLAGS_riscv32imc_esp_espidf = None
  cargo:rerun-if-env-changed=TARGET_CFLAGS
  TARGET_CFLAGS = None
  cargo:rerun-if-env-changed=CFLAGS
  CFLAGS = None
  CARGO_ENCODED_RUSTFLAGS = Some(--cfgespidf_time64)
  cargo:warning=Compiler family detection failed due to error: ToolNotFound: Failed to find tool. Is `riscv32-esp-elf-gcc` installed?

  --- stderr

  error occurred in cc-rs: Failed to find tool. Is `riscv32-esp-elf-gcc` installed?
```

I think I installed the riscv toolchain tools correctly, but maybe not. I just followed the instructions here:
https://github.com/esp-rs/esp-idf-template?tab=readme-ov-file#prerequisites

MabezDev, 2026-11-11:
The crate set from your steps builds fine for me on 1.100.0.0, could you retest with that release and reopen if you still run into issues?

closed 2026-11-11 by MabezDev