esp-rs/rust #256: ICE building dagrs 0.4.2 in release mode with rust 1.85.0.0

kyrias, 2025-03-31:
### Reproduction

#### Setup

```
cargo new --lib foo
cd foo
cargo add dagrs@0.4.2
mkdir .cargo
cat >.cargo/config.toml <<-EOF
[build]
target = "xtensa-esp32s3-espidf"

[target.xtensa-esp32s3-espidf]
linker = "ldproxy"

[unstable]
build-std = ["std", "panic_abort"]
EOF
```

#### Reproducing

```
cargo +esp build --release
```

### Meta

`rustc +esp --version --verbose`:
```
rustc 1.85.0-nightly (fc08bef7e 2025-02-18) (1.85.0.0)
binary: rustc
commit-hash: fc08bef7ed8528e4aed2307195529c55fef62f38
commit-date: 2025-02-18
host: x86_64-unknown-linux-gnu
release: 1.85.0-nightly
LLVM version: 19.1.2
```

### Error output

[rustc-ice-2025-03-31T13_43_02-1999824.txt](https://github.com/user-attachments/files/19536123/rustc-ice-2025-03-31T13_43_02-1999824.txt)

kyrias, 2025-03-31:
More specifically the issue seems to have been introduced in 1.81.  It builds with 1.80.

Though before 1.85 the error we get is:

```
rustc-LLVM ERROR: Error while trying to spill A8 from class AR: Cannot scavenge register without an emergency spill slot!
```

thetek42, 2025-04-03:
I encounter the same "non-UTF8 SMDiagnostic" ICE when trying to build my application. For me, the issue only occurs when `lto = "fat"` is set.

torkleyy, 2025-05-14:
This is still an issue when building with 1.86 (`5944fc5cd 2025-04-01`).

```
thread 'lto everything' panicked at compiler/rustc_codegen_llvm/src/llvm/diagnostic.rs:106:10:
non-UTF8 SMDiagnostic: FromUtf8Error { bytes: [85, 110, 100, 101, 102, 105, 110, 101, 100, 32, 116, 101, 109, 112, 111, 114, 97, 114, 121, 32, 115, 121, 109, 98, 111, 108, 32, 46, 76, 224, 94, 95, 24, 3], error: Utf8Error { valid_up_to: 29, error_len: Some(1) } }
```

MabezDev, 2026-11-11:
Duplicate of #270, your repro builds on 1.100.0.0 now but the underlying spill slot bug is still open there.

closed 2026-11-11 by MabezDev