esp-rs/rust #259: Register allocation failure

MabezDev, 2025-05-06:
Repro steps

1) `git clone https://github.com/okhsunrog/s3_st7789_start - b sync`
2) Modify Cargo.toml's `opt-level` to 3
3) `cargo build --release`

ivmarkov, 2025-06-30:
I'm compiling ATM for the stock `esp32` with `opt-level = 2` (because, esp-storage requires that on `esp32`) and both xtensa 1.86 and 1.87 fail when compiling the `domain` crate with:
```
rustc-LLVM ERROR: Cannot select: 0x7f83540233d0: v4i1 = BUILD_VECTOR Constant:i32<0>, Constant:i32<0>, Constant:i32<0>, Constant:i32<0>
  0x7f8354344e50: i32 = Constant<0>
  0x7f8354344e50: i32 = Constant<0>
  0x7f8354344e50: i32 = Constant<0>
  0x7f8354344e50: i32 = Constant<0>
In function: _ZN6domain4base3opt6subnet15addr_apply_mask17h7378d07d37a38eabE
```

Related to your bug report, or shall I open a separate one?

(Can't try with the older 1.85 which _I think_ used an earlier LLVM, because the ESP baremetal echosystem had moved and wants 1.86 now...)

MabezDev, 2025-07-03:
My guess is that somehow llvm is trying to use vector instructions in the subnet apply mask function, and it's found a vectorization method using bitflags (xtensa has these, mostly for float operations), maybe if you hand roll the parsing or change some of the opts you can get it to work for now.

bugadani, 2025-07-03:
> I can hand that directly to the LLVM team.

Might be better to open an issue directly in https://github.com/espressif/llvm-project/issues so that the team gets the ticket and it stays public, too.

MabezDev, 2025-10-03:
I'm not able to reproduce the og test case with 1.90. @ivmarkov please try your test case when you get a chance.

Closing.